dresden-elektronik/deconz-rest-plugin #2223: Phoscon - setting new password after GW reset leads to Login failed.

mvasicek, 2019-12-16:
Hi, I performed GW reset and setting new password leads to Login failed message. No other way to move further. Device power cycled, docker container ran fresh.

mvasicek, 2019-12-17:
Issue solved by clearing browser's cache.

manup, 2019-12-17:
>Issue solved by clearing browser's cache.

Glad it works now.

Small side note, next year we'll introduce a new login page which will fix a bunch of issues like this one.

increddibelly, 2020-01-22:
is it too soon to talk about an expected update? I keep having this problem again and again to the point that it's just unusable.

closed 2020-05-30 by stale[bot]

SHxKM, 2021-01-11:
I'm also experiencing this issue. Any updates?

ttrueten, 2021-02-14:
I'm also interested in a solution. Reset Link doesn't work. Deleting zll.db and restoring of backup is ok, but seems to overwrite the new password with the forgotten one...

:-(