dresden-elektronik/deconz-rest-plugin #2225: Please fix Issue 1962

Openhabber, 2019-12-18:
Thank you.

joshhodgson, 2019-12-21:
Hi, Openhabber. [Here](https://opensource.com/article/18/1/step-step-guide-git) is a guide on how to use git. If you're having an issue with open source software, try and fix it yourself. The developers of this software don't owe you anything.

Happy holidays 🎄

Openhabber, 2019-12-21:
Im sorry,
Dresden Elektronik is a company selling zigbee hardware commercially and phoscon/dexonz is thw software for that, proposing it is a product that works and is commeecially developed, or am i wrong in that regard? 
https://www.dresden-elektronik.de/funk/software.html

Open source just means they supply the code to use with a normally more open license.
I think you misunderstand open source here. 
It is still a commecial product which is advertised as necessary for the hardware you can buy. 

This is not a communiry product of some hobby programmers. I paid for their hardware and expect it to work like advertised, or at least care about bugs and THEIR Software.

kasteleman, 2019-12-21:
The hardware itself works. You are suffering from a firmware issue of a
other manufacturer product. Other manufacturers have same sort of issues.
Some can be fixes (read workaround applied) others take some time and there
are those that can’t be fixed at all! Don’t blame Dresden for that! They
are doing the best they can!

ebaauw, 2019-12-21:
> I paid for their hardware and expect it to work like advertised, or at least care about bugs and THEIR Software.

Fair enough, but then please contact dresden elektronik customer support.  GitHub _is_ supporting a community of developers and is not really suited for providing end-user support.

joshhodgson, 2019-12-21:
> Dresden Elektronik is a company selling zigbee hardware commercially

Their hardware isn't the issue here, nor is the connection to your bulbs. The issue is that your lightbulbs are misbehaving and not doing what they're told, no?

manup, 2019-12-21:
@Openhabber I've described the current state of the problem in the origin issue.
I close this issue, please continue conversion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1962

closed 2019-12-21 by manup