dresden-elektronik/deconz-rest-plugin #2226: Devices Offline should show as unavailable

greencharlie100, 2019-12-19:
Devives that are off the network should show as unavailable in the APP. 
Can this be fixed?

Example. I have a Zigbee smart plug which is paired and working as expected. Then I switch off the mains power -  meaning the Zigbee smart plug is no longer connected to the network. It still shows up as available (switchable) in phoscon etc...

This is a problem.

Smanar, 2019-12-19:
???
I have same, and the device is grayed in phoscon, a bulb and a plug. just need some time for deconz mrk them as unrechable.

wvuyk, 2019-12-19:
It will go unreachable, but it takes a lot of time. It seems to be a bit random, here a INNR plug takes at least 6 hours before showing offline. A Philips bulb here can take more then a day and a xiaomi open/close sensor never became unreachable.

Devices becoming unreachable is not really trustworthy in common for my experiances too.

Smanar, 2019-12-19:
Ha yes, I understand.
And yes, I m agree with you, It's hard for batterie sensor, but I don't understand for bulb as the reporting is every 300s for exemple, a bulb without reporting for more than 5mn is not reachable for me, don't need so much time.

Perhaps there is a reason ?

manup, 2019-12-20:
There are various factors which play a role here:
* A simple rather short timeout is tricky when considering scaling for large networks
* Some devices support reporting some need to be polled
* Some end-devices don't report periodically but only when for example a button is pressed (Trust, Jung, ...)

There is plenty of room for improvement but it's a challenging task. I think with the v2 API this very aspect of determining the reachable state should get a clean design which addresses the differences in devices. Perhaps we can add a default behavior which can be customized with per device/vendor configuration files (JSON or XML).

greencharlie100, 2019-12-20:
I guess if mains powered devices were marked as unreachable within 1-3 minutes would be good. As for battery-powered sensors etc... I am not overly concerned with those. If they were described as unreachable anytime up to 60minutes after the last contact with the server would be perfect. 
My issue normally exists when I try and control a device (light, switch etc) when it is actually powered off at the mains. this is a scenario that makes a smart home appear as actually dumb.

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored, see #2590.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix