dresden-elektronik/deconz-rest-plugin #2228: 2.05.69+ versions of deconz looses connections with IKEA lamps

PremiumUsername, 2019-12-21:
Deconz has been amazingly stable for me for the past year, with everything connected to it never losing the connection even once. This stability came to an end when I upgraded to 2.05.71 and also with 2.05.72. With those versions my IKEA TRADFRI bulbs (GU10 W 400lm, FW Version 1.2.214)  have been loosing their connection multiple times. The only thing that makes them come back is powering them off and on again. 

To summarize, in over a year everything has been working 100%, but with the new version I've lost the connection 4 times with 3 different lamps (of that same type) within a couple of days. I restored an old backup (2.05.69) and now everything is working flawlessly again.

I'm using the original conbee with an RPI 3+ Hass.io installation.

PremiumUsername, 2020-01-01:
Just to update, running on 2.05.69 now for 10 days, and it is totally stable. Clearly there is something different with the 2.05.69+ versions when it comes to interoperability with some lamps.

jtebbens, 2020-01-04:
I have to agree here, I also see a difference between 2.05.71 and 2.05.69 with Tradfri sockets and Innr bulb being unavailable. This is with a conbee II stick btw.

alex2000-ha, 2020-01-08:
i am going crazy every day an other device is missing. conbee 2 2.05.72
Xiaomi contact sensors, Eurotronic thermostats, ...

jtebbens, 2020-01-08:
> i am going crazy every day an other device is missing. conbee 2 2.05.72
> Xiaomi contact sensors, Eurotronic thermostats, ...

Try 2.05.69, that made a huge difference

PremiumUsername, 2020-01-08:
I feel I have to make a cautionary note here. I assumed this problem had something to do with deconz because the light went missing in Phoscon, but I was using Home assistant and the deconz plugin for HA in order to run Phsocon. In the most recent HA plugin (V 5.0), I'm cautionary optimistic that things have changed for the better even tough the deconz version used is still the "problematic" one. It's only been a couple of days with the new HA plugin version but so far so good. 

Were you using Hass.io, in order to run deconz?

jtebbens, 2020-01-08:
Nope, no hass.io here. Using an custom app with Homey so it talks directly to deconz

PremiumUsername, 2020-01-17:
> I feel I have to make a cautionary note here.

Cautionary note withdrawn , now I've lost contact with one lamp again. Something is broken in  the new versions.

jtebbens, 2020-01-17:
@PremiumUsername do you use group actions for those lamps? I read/understand Ikea lamps seems to act better in group actions rather than single on/off actions per lamp.

PremiumUsername, 2020-01-17:
@jtebbens I use both group actions and single, but once a lamp is "lost" it is unavailable for both types of control. Can you still control them with group commands?

closed 2020-05-23 by stale[bot]