dresden-elektronik/deconz-rest-plugin #2229: Busch Jäger wallswich 2 and 4 channel only recognized as 1 channel

Wegatarus, 2019-12-21:
I try to connect the busch jäger wall switch  6736/01-84 and  6737/01-84 (2 and 4 channel version).
Always if  the Phoscon App connect the switch it is showed as a 1 channel swtich. I use the Version 2.05.71 Firmware 26330500.
How can I connect the multi channel switchs?
Thanks

SwoopX, 2020-01-03:
Can you be a bit more precise? Mine is working fine. Did you check via the switch editor in Phoscon? Have you checked the sensors via REST API?

Wegatarus, 2020-01-03:
I use the switch editor in Phoscon and I try my 1,2 and 4 channel switchs. Always if Phoscon connecting the new switch it shows only the 1 channel version. Then I can define the  switch bottoms with lights but it does not work.

SwoopX, 2020-01-03:
ok. Then, I'd recommend trying to re-pair the switch. If I recall correctly (it's been a year or so), my first attempt ended similarly.

Another option could be start sensor search in Phoscon and read the simple descriptors in deconz GUI. Alternatively, start pushing the remaining buttons while search is still active.

Wegatarus, 2020-01-03:
Thanks, I will try it again. But I use the headless version and have no idea about the deconz GUI. Do you know about any device id to identify the devices? If it only reconnized the 1 channel switch it must be the id wrong?

SwoopX, 2020-01-03:
Not sure if it can be done without deconz GUI. What hardware are you running and do you have an X server available? Otherwise, check the wiki and look for "Access deCONZ GUI in headless setups"

WarmCatUK, 2020-03-18:
Also chiming in here with exactly the same issue. It seems that Busch-Jaeger switches are not at all compatible with deCONZ and raspbee.
I add the 4-channel Transmitter in the "Add new switch" section. The switch is detected, however only as a "wall mounted transmitter 1-channel" Version 1.0.4. Obviously it is not possible to map the buttons to anything, anyone with experience of these switches knows that you have to scan network with the switch after adding it to a network, then press and hold each switch individually to save the scenes.
I would suggest that these switches are removed from the compatible switch list.

SwoopX, 2020-03-18:
Have you thoroughly followed the instructions displayed for joining? Which deconz version are you running? Sometimes, just waiting for some minutes helps or a deconz restart...

It is btw compatible, running it for over a year now without any issues.

Larx, 2020-03-23:
Just wanted to chime in that I have exactly the same problem. Not much more I can do to help.

Larx, 2020-03-25:
Is there any way to push this issue to the manufacturer? I'm in the process of equipping my flat with Busch-Jäger stuff, and both Busch-Jäger and Dresden Electronic list this equipment as supported. Which it isn't (or some change in the last months broke the support, as it seemingly worked some time ago).

This is very annoying and it means that an advertised capability concerning the support of a major brand is not working properly.

WarmCatUK, 2020-03-25:
@SwoopX 
I contacted Busch-Jaeger directly and it’s confirmed that the wall mounted transmitters are not compatible unless the firmware is upgraded to 1.2.0
I have been quoted £96.65 for the ZLL USB Update Stick (part no 2CKA006710A0009) with a two week lead time.
It is not possible to update over the air from deCONZ when the switches are at version 1.0.4
Hope this info helps the others who are having problems; I’ll update in a month when I get back home from work.
Assuming it works, I’m happy to update switches for anyone if they pay postage.

Larx, 2020-03-25:
Thanks for the information. To be sure, I would have expected a hint like this on the official homepage, as it seems to be a known limitation :-( .
I maybe also would contact you because of the update if I don't find a local possibility.

SwoopX, 2020-03-26:
Obvisously, you do not seem to be interested in any resolution you experience here since all I read is about expectations and disappointment. Also, I do not recognize any details about your current setups and technical details, maybe I just overlooked them. Additionally, I do not get the impression that you used the search function. I may recommend to come back to a reasonable discussion if you're really interested in any resolution and enabling people to help you.

Btw, I run the 4-gang and it works without any issues.

Larx, 2020-03-26:
@SwoopX: Thanks for your neutral, helpful and friendly contributions to this bug report. I will not react to any further comments from you.

Larx, 2020-03-26:
I contacted the German Busch-Jäger support with the information given by @WarmCatUK in order to find out if there is a service solution apart from buying a dongle here. Will post if I get a result.

Larx, 2020-03-27:
Update: The BJ support gave me the address of a local service representative which should be able to do the necessary update for a nominal fee. I will try this when times are better :-/ .

Larx, 2020-03-31:
So I had the opportunity to use the ZLL update stick (BJ 6719). After the update the wallswitch works as intended. A few caveats:

1. Software is Windows-only. Stick and software worked flawlessly on my Linux box using my old Windows 7 VMM, however.
2. I used the variant to open a new network from the wallswitch (5 sec. both topmost buttons, then right button).
3. The update process never stops. The software seems to keep on scanning the network for updateable devices. But as soon partial progress bar for the device you want to update has reached 100%, it seems to be safe to stop the process. (I grew up with the dogma "never stop a running flash upgrade unless you want to brick your device", so I at first hesitated to hit the "Cancel Update" button prematurely ;-)
4. Disconnect the wallswitch after the update process and leave it powerless for a few minutes. At first I didn't do that, only software-resets, and then the 4 channel wallswitch only appeared as 2 channel wallswitch in the Phoscon app.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix