dresden-elektronik/deconz-rest-plugin #2231: Ikea Tradfri On/Off Switch

nicedevil007, 2019-12-22:
I've got multiple IKEA Tradfri switches (that one with an upper and lower button, not the big round one).

In Node-RED I can use the buttonevent 1001/1003 and 1002 + 2001/2003 and 2002.
In the Phoscon App I can only setup 1002 and 2002.

D3nnisd, 2019-12-23:
This switch behaves differently anyway. I use it to make the Fyrtur go up and down (just with the open cover command) within HAS. But going up makes it go for 1 second and after that it is stuck. A Philips remote functions as intended.

nicedevil007, 2019-12-29:
Mhhh maybe someone can solve this prob? :)

allanpersson, 2020-01-18:
Any plans to implement this in the app under switches?

manup, 2020-01-18:
Yes, the whole switch support as in integration wizards and switch editor support will be largely extended.
We're working on it since a few months now, can't provide an ETA yet other than during the next months.

allanpersson, 2020-01-18:
@manup awesome news 👏 It's just much easier with the stuff in the UI 😉
Thank you for the great work your doing 🙏

closed 2020-05-24 by stale[bot]