dresden-elektronik/deconz-rest-plugin #2232: could not sync with bootloader

Christian-Kauth, 2019-12-22:
I tried to flash the bootloader, because the red led is blinking permanently. I use Raspberry Pi 4B.

Updating WiringPi:

`chris@RASP-4B-4GB-SERVER-2:/tmp $ wget https://project-downloads.drogon.net/wiringpi-latest.deb
--2019-12-22 15:58:25--  https://project-downloads.drogon.net/wiringpi-latest.deb
Resolving project-downloads.drogon.net (project-downloads.drogon.net)... 188.246.205.22, 2a03:9800:10:7b::2
Connecting to project-downloads.drogon.net (project-downloads.drogon.net)|188.246.205.22|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 52260 (51K) [application/x-debian-package]
Saving to: \u2018wiringpi-latest.deb\u2019

wiringpi-latest.deb 100%[===================>]  51.04K  --.-KB/s    in 0.1s    

2019-12-22 15:58:25 (486 KB/s) - \u2018wiringpi-latest.deb\u2019 saved [52260/52260]

chris@RASP-4B-4GB-SERVER-2:/tmp $ sudo dpkg -i wiringpi-latest.deb
(Reading database ... 166454 files and directories currently installed.)
Preparing to unpack wiringpi-latest.deb ...
Unpacking wiringpi (2.52) over (2.52) ...
Setting up wiringpi (2.52) ...
Processing triggers for man-db (2.8.5-2) ...
chris@RASP-4B-4GB-SERVER-2:/tmp $`

`chris@RASP-4B-4GB-SERVER-2:/tmp $ gpio -v
gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 01, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
chris@RASP-4B-4GB-SERVER-2:/tmp $`

`chris@RASP-4B-4GB-SERVER-2:/tmp $ wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_Rpi-latest.bin.GCF
--2019-12-22 16:02:09--  http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_Rpi-latest.bin.GCF
Resolving deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 84.19.195.242
Connecting to deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|84.19.195.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 125802 (123K) [application/x-graphing-calculator]
Saving to: \u2018deCONZ_Rpi-latest.bin.GCF\u2019

deCONZ_Rpi-latest.bin.GCF       100%[=====================================================>] 122.85K   387KB/s    in 0.3s    

2019-12-22 16:02:10 (387 KB/s) - \u2018deCONZ_Rpi-latest.bin.GCF\u2019 saved [125802/125802]

chris@RASP-4B-4GB-SERVER-2:/tmp $`

`chris@RASP-4B-4GB-SERVER-2:/tmp $ sudo GCFFlasher_internal -f deCONZ_Rpi-latest.bin.GCF
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
action: update firmware after 2 ms
could not sync with bootloader
chris@RASP-4B-4GB-SERVER-2:/tmp $`

`chris@RASP-4B-4GB-SERVER-2:/tmp $ sudo GCFFlasher_internal -r
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
chris@RASP-4B-4GB-SERVER-2:/tmp $`

The status LED is green a short time an then it is blinking again red permanently.

Can anyone help?

Christian-Kauth, 2019-12-22:
solved

in added in /boot/config.txt 

dtoverlay=pi3-disable-bt

closed 2020-04-30 by stale[bot]

hmrc87, 2020-05-09:
Same problem here but on RPI4. Disabling BT via dtoverlay=disable-bt in /boot/config.txt does NOT help :(

Smanar, 2020-05-09:
Hello, same problem ?
Raspbee 1 too ? It have never worked ? You have set the serial port and used the good wingripi version ? https://phoscon.de/en/raspbee/install#raspbian

Nemesis24, 2020-05-13:
The same for me

Smanar, 2020-05-13:
@Nemesis24 
>Hello, same problem ?
Raspbee 1 too ? It have never worked ? You have set the serial port and used the good wingripi version ? https://phoscon.de/en/raspbee/install#raspbian