dresden-elektronik/deconz-rest-plugin #2233: Moving lights from Hue Bridge to Deconz

kmplngj, 2019-12-22:
Hello,

this is more an ask for help than an issue.

I tried to find the best way to move my lights from the hue bridge to Deconz.

I found some informations about Touchlink, which I could not do in my setup. And I found some info regarding hue lights resets with a hue dimmer. But this did not work for me.

Could someone write a Wiki entry to describe the process in detail?

Thanks
Jan

Smanar, 2019-12-22:
It works https://www.youtube.com/watch?v=qvlEAELiJKs
You can too without remote, video available on same site.

Kane610, 2019-12-22:
Make sure the light is removed from all rooms groups remotes and then remove it from hue. Then it would just be to pair it

closed 2019-12-24 by kmplngj