dresden-elektronik/deconz-rest-plugin #2234: Ubisys D1 ZCL command "Move to level" :bulb: fails 50%

o-live-r, 2019-12-23:
I am looking for a working solution for programming the two switch inputs.
My goal is that switching input 1 dims up to 20%:bulb: when switched on via a static switch (toggle switch) and dims down to 0%:bulb: when switched off.
Switching input 2 should dim up to 100%:bulb: when switched on and dim down to 20%:bulb: when switched off.

The function is to use a twilight switch at switch input 1 to switch on a night light and when a motion detector is triggered, the light at switch input 2 is to be raised to 100%:bulb:.
As soon as the run-on time of the motion detector has expired, the light should then be dimmed back down to 20%:bulb:. The twilight switch is to switch off the light completely dimmed at a certain brightness. The motion detector has also integrated a twilight switch and will then no longer trigger even with the same brightness. All switching operations should be dimmed.

All of this should work without a zigbee connection and be programmed in advance on the D1.
I have already tested the following ZCL commands with 50% success rate :disappointed: :

Status | 4841 |   |   | element type: 0x41 (raw data)
-- | -- | -- | -- | --
  | 04 | 00 |   | element count: 0x0004 (4 entries)
works | 10 |   |   | element #1: ten bytes
works | 00 |   |   | InputAndOptions: 0x00
works | 0D |   |   | Transition: released -> pressed
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on   D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | FF0E | 0000 | ZCL Command Template: Move to level, transitiontime,
fails | 10 |   |   | element #2: ten bytes
fails | 01 |   |   | InputAndOptions: 0x01
fails | 0D |   |   | Transition: released -> pressed
fails | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on   D1)
fails | 08 | 00 |   | Cluster ID: 0x0008 – level control
fails | 04 | FF0E | 0000 | ZCL Command Template: Move to level, transitiontime,
works | 10 |   |   | element #3: ten bytes
works | 00 |   |   | InputAndOptions: 0x00
works | 03 |   |   | Transition: any state -> released
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on   D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | 2A10 | 0000 | ZCL Command Template: Move to level, transitiontime,
fails | 10 |   |   | element #4: ten bytes
fails | 01 |   |   | InputAndOptions: 0x01
fails | 03 |   |   | Transition: any state -> released
fails | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on   D1)
fails | 08 | 00 |   | Cluster ID: 0x0008 – level control
fails | 04 | 2A10 | 0000 | ZCL Command Template: Move to level, transitiontime,

Unfortunately I can't get any further without help, because I don't understand why calling the "Move to level" function for input 0x00 works for both switching operations at switch 1 and the function completely fails for input 0x01 (switch 2).
The values ​​from the table only describe my approach. The above-mentioned goal remains. Unfortunately I did not find any further explanations in the technical reference. Where is the mistake?:thinking:

ebaauw, 2019-12-23:
`10` is sixteen bytes, not ten.  If I count correctly, the second element is taken to be four bytes in length and the third element happens to start where you think it would.

o-live-r, 2019-12-23:
You got me on the right track. I experimented with the byte length and with 9 bytes it worked just fine. :smiley: :+1:

o-live-r, 2019-12-23:
Here is the right code. I forgot to mention that I edited the length of the ZCL template.

Status | 4841 |   |   | element type: 0x41 (raw data)
-- | -- | -- | -- | --
  | 04 | 00 |   | element count: 0x0004 (4 entries)
works | 09 |   |   | element #1: nine bytes
works | 00 |   |   | InputAndOptions: 0x00
works | 0D |   |   | Transition: released -> pressed
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | FF | 0E00 | ZCL Command Template: Move to level, transitiontime,
works | 09 |   |   | element #2: nine bytes
works | 01 |   |   | InputAndOptions: 0x01
works | 0D |   |   | Transition: released -> pressed
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | 2A | 0E00 | ZCL Command Template: Move to level, transitiontime,
works | 09 |   |   | element #3: nine bytes
works | 00 |   |   | InputAndOptions: 0x00
works | 03 |   |   | Transition: any state -> released
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | 2A | 0100 | ZCL Command Template: Move to level, transitiontime,
works | 09 |   |   | element #4: nine bytes
works | 01 |   |   | InputAndOptions: 0x01
works | 03 |   |   | Transition: any state -> released
works | 02 |   |   | Source: Endpoint #2 (hosts primary on/off client cluster on D1)
works | 08 | 00 |   | Cluster ID: 0x0008 – level control
works | 04 | 00 | 0100 | ZCL Command Template: Move to level, transitiontime,

closed 2019-12-23 by o-live-r

wb441, 2022-02-16:
Dear o-live-r,

I was so happy to find this thread. Been looking for a way to do this!
But I'm missing a crucial piece of information: how/where do you send this configuration to the device?

Best regards,

o-live-r, 2022-02-16:
see the comment of sheineme commented on 9 Jan 2019 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/360#issuecomment-452850774