dresden-elektronik/deconz-rest-plugin #2236: Ping/Pong missing in Websocket implementation

doodieb15, 2019-12-23:
RFC 6455 describes a mechanism to check whether a connection is still alive (ping/pong). The QT Websocket class also includes this mechanism. However, it is not implemented in the deconz REST/API, which is unfortunate.
Please include this feature in the deconz REST/API, it would help to make a robust websocket connection through unreliable channels.
Thanks

dergreg, 2019-12-27:
@ebaauw Could this also be useful for [https://github.com/ebaauw/homebridge-hue](homebridge-hue) or is the plugin notified early enough of a disconnect?

ebaauw, 2019-12-27:
I use the [`ws`](https://github.com/websockets/ws) package to handle the web socket.  According to the [documentation](https://github.com/websockets/ws#how-to-detect-and-close-broken-connections), it supports ping/pong, but I never used this (at least consciously).

I haven't seen any issues with homebridge-hue nor with `dc_eventlog`.  Both reconnect automatically, when the web socket is closed.  I haven't seen any delays between deCONZ stopping and the web socket reported as closed.  I tend to run deCONZ and homebridge on the same Raspberry Pi, using the loopback interface.  I suppose that's the most reliable channel you can get.

Occasionally, I run homebridge on a different machine (mostly when testing changes to my plugins), but haven't seen anything funny either.  I haven't done any extensive testing, like pulling network cables or rebooting the network switch or router.

homebridge-otgw also uses the web sockets exposed by the OpenTherm Monitor.  I haven't seen anything funny there either.  But the again, I run homebridge on the same Pi as the OpenTherm monitor.

doodieb15, 2019-12-27:
As long as the channel is reliable,everything is fine. I am running the websocket through an unreliable VPN channel and sometimes the client gets disconnected.Without the ping pong feature the client has no possibility to find out and reconnect if needed.

Jey-Cee, 2020-02-07:
I've seen the same issue. After a random time the websocket connection seems to be still alive, but there comes no data. 
There is no message for close or error, so it is not possible to detect this situation.

Mimiix, 2020-06-05:
It seems this issue is resolved. If it is now, please re-open!

closed 2020-06-05 by Mimiix