dresden-elektronik/deconz-rest-plugin #2237: DeConz app not showing any content

Koal4, 2019-12-23:
Hi,

I hope someone can help me with the following issue. I've been trying to switch to HomeAssistant and i've stumbled upon some problems with this conbee stick (version 1) which happened to me about 3 times now. If I reinstall Homeassistant and pair the conbee stick it works untill the following happens.

I was able to pair Ikea lights and Aqara sensor succesfully. I also had them show up in HA so far so good. But since I'm in the process of learning HA I sometimes reboot my raspberry pi 3b. And a couple of time i just loose connection with the stick. I open DeConz app from HA and all content (like paired bulbs + sensors) are gone and even the gateway page is blank.

BUT i'm still able to control my bulbs + sensors from withing HA. No matter what I do and how many times I reboot I won't get anything back. If I reinstall HA and deconz it works fine again for a couple of hours / reboots. 

[image]
[image]

Smanar, 2019-12-23:
Hello,
Have you tried with another browser ? (To check cache problem)

Koal4, 2019-12-23:
Hi,

I think you're on to something because in the other browser I did experience the same but after that I cleared out the cache and suddenly it came back in my working browser (chrome)

closed 2020-04-30 by stale[bot]