dresden-elektronik/deconz-rest-plugin #2238: IKEA Leptiter built-in spots not showing up

Qonstrukt, 2019-12-23:
I've had some good success lately in adding all my Tradfri lights to a RPi 2 with a Raspbee. All my lights, sensors and switches are showing up, except for 6 Leptiter built-in spots. They don't even show up in deCONZ app, not even as unrecognised devices.

They did work correctly with the IKEA hub, and for now I've paired them directly with an IKEA motion sensor which also works as it should. But it's weird they're not even showing up in deCONZ itself right?

Kinda stuck from here on what else I can try.

SwoopX, 2020-01-03:
Have you checked the wiki if the light is supported? If not, you may want to issue a device support request.

Qonstrukt, 2020-01-03:
Yes, but that’s the point. It’s not even visible, so I can’t do a new device request. 😄

ebaauw, 2020-01-03:
I've got no experience with the Leptiter, but from the online manual, they should work like any other IKEA Trådfri light.  Some general suggestions:
- Make sure to power cycle the spot after the reset, while searching for new devices.
- Did the spot fully reset?  The timing of the ON/OFF is somewhat delicate, at least for the IKEA Trådfri lights I have.
- If you have a Hue dimmer switch, you might try and reset the spot via touchlink (holding the _On_ and _Off_ buttons for 10 seconds while keeping the dimmer close (< 10 cm) to the spot).
- Did you build in the spots already?  Could cause radio interference.  Try pairing the spot closer (< 1m) to the Raspberry.  If you cannot move the spot, power the Raspberry from a USB power bank to make it portable.
- You might also try a touchlink reset from the old web app.  Menu|Settings|Scan for Devices, while keeping the spot and Raspberry close to each other.

Qonstrukt, 2020-01-03:
Thanks, I was wondering if those options were feasible if you can’t even see them after reset. (I’m 99% sure it was done properly.)
But I’m gonna try some of those options. My dad has a hue dimmer, so I’ll borrow that one and see how that goes.

Qonstrukt, 2020-01-08:
I'm not sure what I'm doing wrong, but it starts to feel like my (bad) experience from a year ago again. I had to replace a GU10 lamp and I cannot seem to find any new devices at all anymore.

I'm able to reset devices through TouchLink in de old web app and with the Hue dimmer remote, but adding new devices seems impossible. 

Why is this happening? Is there some kind of limit to the max amount of devices? I have 29 nodes now, of which all but 9 are routers. Whether it be GU10 or E27 bulbs, or the Leptiter ones, deconz just doesn't show any new devices any more after opening the network.

filippocamillo, 2020-01-09:
Hello there! 
I have the very same issue. I'm new here but I'm not able to continue my setup. The only devices which are currently working are the IKEA remote control and the motion sensor.

Thanks for your help!

Qonstrukt, 2020-01-09:
After skipping a day and retrying to pair the new GU10 light, I got that going again, so far so good. Feels a bit flaky though, but it works.

The Leptiter ones though...
I'm able to pull them from the ceiling a bit, but I really can't get them to show up in my network in the Deconz application. The weird part is that I'm able to reset them, not only with the Hue dimmer, but also with the Touchlink reset in the old web app. I can even make them blink!
After a reset they show up on another channel than I'm currently using (20 vs 11), would it help to mess with the Channel mask settings for my network in Deconz? I'm kind of afraid to touch any settings there.

ebaauw, 2020-01-10:
> After a reset they show up on another channel than I'm currently using (20 vs 11)

Do you mean in the touchlink overview of the old web app?  What PAN ID do they report?  Do you know that PAN?  Could one of your neighbours have a permanently open ZigBee network, causing the light to join the wrong network?

Try touchlinking a the light paired to the Trådfri hub, to find out the PAN ID of the hub.

Qonstrukt, 2020-01-10:
I'll gather some more information this weekend, I really appreciate your help with this. 😅 
I don't have my Tradfri hub in use anymore, but if it's needed I could pull it out again. Pairing my lamps with the Motion Sensor doesn't give any problems though, and I can see all paired devices on a single network with the sensor. (It was channel 18 I think.) Then resetting them gave them channel 20, that's what I can remember from the top of my head. I don't know if they then just pick a random channel when resetting? I will look for the PAN ID's when I get to it. 👍

Qonstrukt, 2020-01-15:
Ok, so I tried Touchlink scanning some more, I picked a Leptiter, and it has the following ID: 0x000d6ffffe1a2248

My network is on channel 11, with ID 0xA29B
The Motion sensor is on channel 20 with ID 0xAB40
The lamp when Touchlink reset: 25 with ID 0x2606

What does this tell exactly? After reset it creates it's own network I guess? I don't think it's adopted by another coordinator because it stays at 100% the whole time.
Is there anything I could do to help the pairing process? I still can't have it join my own network, not even with the Raspbee 10cm away from it.

Qonstrukt, 2020-01-15:
Wow, I actually got one to show up now! But only doing the IKEA reset (6x ON/OFF) while having the Raspbee very close by at the same time. So this is gonna be a challenge, but I finally have the details for them to be added:

[image]

SwoopX, 2020-01-15:
Sounds good. Can you please also paste a screenshot of the node info panel?

Qonstrukt, 2020-01-15:
[image]

I'm starting to notice a pattern. I accidentally just reset 3 spots I have. 
Two of them are LED1650R5, and I had to replace another which is now a newer model LED1837R5.

When I open my network, and add the two LED1650R5's, they show up in a matter of seconds. When I try to add the LED1837R5, I can't for the life of me get it to show up, just like the Leptiter ones. Resetting it 10 times, moving the Raspbee up against it, do a voodoo and a rain dance eventually makes it appear, but it feels like I just ran a marathon.... 🙁 

I seems like the newer IKEA lamps are much less prone to join open networks? Or at least one coordinated by a Raspbee. It might be anecdotal, but this is what I notice.

Qonstrukt, 2020-01-15:
To finish off the night, I did a run of deCONZ with debugging turned on as per the wiki. Touchlinked a Leptiter, opened the network and kept the Raspbee close, but couldn't see the device ID pass by in the logging. Really annoying lamps so to say.

SwoopX, 2020-01-17:
If the last screenshot was from the newer model, it may have something to do with IKEA devices using yet unknown MAC prefixes belonging to those. That's already addresses by a merged commit so there is a certain chance that this challenge may be resolved with the upcoming release.

Qonstrukt, 2020-01-18:
The last screenshot is from the Leptiter that you asked for 😊

jensflorian, 2020-03-05:
Does the latest Deconz now recogize the LEPTITER spots instantly? Just asking before picking up some of these.

ebaauw, 2020-03-05:
Lights are recognised by _Profile ID_, _Device Type ID_ and _Receiver on when idle_ (from the descriptors), not by mac address, _Model ID_, and _Manufacturer Code_.  That’s why lights don’t need to be whitelisted.

I see the light advertises a ZHA profile, but a _Color Temperature Light_ device type.  Can some-one sniff the light’s descriptors and double-check the device ID: is that the ZLL device id, or the ZB3 one?  In the first case: the current code won’t recognise the light.

Qonstrukt, 2020-03-05:
> Does the latest Deconz now recogize the LEPTITER spots instantly? Just asking before picking up some of these.

It will recognise them, but you will need to hold the RaspBee or ConBee **really** close, and have no interference at all.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

excedra, 2020-07-04:
> [image]
> 
> I'm starting to notice a pattern. I accidentally just reset 3 spots I have.
> Two of them are LED1650R5, and I had to replace another which is now a newer model LED1837R5.
> 
> When I open my network, and add the two LED1650R5's, they show up in a matter of seconds. When I try to add the LED1837R5, I can't for the life of me get it to show up, just like the Leptiter ones. Resetting it 10 times, moving the Raspbee up against it, do a voodoo and a rain dance eventually makes it appear, but it feels like I just ran a marathon.... 🙁
> 
> I seems like the newer IKEA lamps are much less willing to join open networks? Or at least one coordinated by a Raspbee. It might be anecdotal, but this is what I notice.
> 
> This newer spot also has the Green Power GP Proxy Basic block attached btw, that is what unites those difficult to pair lamps.

About the LED1837R5: I also have 4 of them. No chance to even reset them (6 times on/off). Simply no interaction. The Pi is directly placed on the Lamp (with an powerbank), so distance should be no issue. 
Fresh once (directly out of the box) do also not start a pairing process on the first power on.

I'm going to try this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1089
But first I need to buy an Philips Dimming Remote for that...

excedra, 2020-07-07:
So, tried the reset method with a Philips Dimming Remote. 
The bulb resets (dimming/flickering) but does not pair.

How to solve this?

Mimiix, 2020-07-07:
Could you open a seperate bug report @excedra ?