dresden-elektronik/deconz-rest-plugin #2240: Unable to update the firmware of Conbee II to 26490700

gabrielemariotti, 2019-12-24:
I am using the Phoscon App with a Conbee II through the Home Assistant/Hassio addon.
The device is connected to `/dev/ttyACM0` port and works fine.

I am running the 2.05.72 with the firmware 26420700.
In the log I can see:

```
11:25:30:068 GW firmware version: 0x26420700
11:25:30:068 GW firmware version shall be updated to: 0x26490700       
```

In the UI I can run the update and it ends successfully.
However when I open again the UI the firmware is still 26420700 and it suggests again the update to 26490700 .
Tried many times, restarting the addon but nothing changed.

th3cube, 2019-12-24:
pls follow these steps to update the Firmware in a Docker Conainer 

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

gabrielemariotti, 2020-01-14:
Sorry for the delay.
Can I run this command also in a raspberry with hassio + home assistant? I am using a specific addon for this and I didn't install it directly with a docker image.

gabrielemariotti, 2020-01-14:
Sorry for the delay.
I am not sure that I can run this script using the addon provided by hassio-home assistant.

closed 2020-05-20 by stale[bot]