dresden-elektronik/deconz-rest-plugin #2243: IKEA Tradfri Remote(FW  2.3.014) not pairing completely

hmrc87, 2019-12-25:
Hi,
I tried pairing  a Ikea Tradfri remote (round) with FW  2.3.014 with Phoscon App and Gateway version 2.05.71 (2019-11-17). 

It finds the device, generates an entry in the DB (as far as I can tell from the tailing the deconz logs) but when the last step appears where I have to push the middle button for 10seconds I cannot get further. The command is not recognized and after a few minutes it shows me INSTALLATION FAILED, TRY AGAIN.

However I can see the tradfri remote in Phoscon, I can use it to control groups but it does not work. No button click does anything.

Is it a problem with the new firmware? If so: How can I downgrade the tradfri remote or when do you plan to upgrade?

ginkel, 2019-12-26:
I am able to pair a TRADFRI remote with FW 2.3.014, so it's most likely not the new firmware revision.

SwoopX, 2019-12-26:
Could you please post a screenshot fo deconz GUI node info panel? What may help you to get the issue resolved can be found in the following two posts:

[#2052](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2052#issuecomment-552202908)
[#1873](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1873#issuecomment-551502335)

hmrc87, 2019-12-27:
[image]
Hi, thats the state after the fifht or sixth pairing try. I deleted the node and then tried it from the phoscon app again.

In some issues there is the "old web app" mentioned. How do I access it? I have the deconz docker container running

SwoopX, 2019-12-28:
Thanks. Just stumbled over it while looking at the screenshot. Eventually, it could be related to a check for supported devices failing. The MAC is a silabs2 and not yet associated to IKEA devices... I'll have a closer look later.

gimini91, 2019-12-29:
Same issue here. 10s press is not recognized during pairing process.

alphahotel29, 2020-02-09:
Same Issue for me. I've just pair two TRÅDFRI remote control. I'm using phoscon gateway. one is on version 1.2.223 and pair without problem. I can assign directly to lights on phoscon app. 

The second one is on version 2.3.014. when i pair it, I see it but i can't enter on the menu on this switch, and assign lights.
When I try to get info via deconz API Rest, i see that first and functional remote create an hidden group, and a sensor.
`  "23242": {
    "action": {
      "bri": 127,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": false,
      "sat": 127,
      "scene": null,
      "xy": [
        0,
        0
      ]
    },
`
the second one just create a sensor:

`  "2": {
    "config": {
      "alert": "none",
      "battery": 60,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "c1b9bXXXXXXXXXXXd4a178",
    "manufacturername": "IKEA of Sweden",
    "mode": 3,
    "modelid": "TRADFRI remote control",
    "name": "Interupteur Salon 2",
    "state": {
      "buttonevent": 4002,
      "lastupdated": "2020-02-09T19:52:51"
    },
    "swversion": "2.3.014",
    "type": "ZHASwitch",
    "uniqueid": "ec:1b:XX:XX:XX-01-1000"
  },
  "3": {
    "config": {
      "alert": "none",
      "battery": null,
      "group": "23242",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "c1663eeXXXXXXXXXb0294c7",
    "manufacturername": "IKEA of Sweden",
    "mode": 3,
    "modelid": "TRADFRI remote control",
    "name": "TRÅDFRI remote control",
    "state": {
      "buttonevent": 1001,
      "lastupdated": "2020-02-09T19:54:45"
    },
    "swversion": "1.2.223",
    "type": "ZHASwitch",
    "uniqueid": "00:0d:XX:XX:X0-01-1000"
  }
}
`

SwoopX, 2020-02-09:
Ok, we're mixing things up here, let's try to get that sorted. From the information provided, I'd conclude that for the device running firmware 2.3.014, paring do go alright (leaving the group issue aside). If yes, good, original issue seems to be solved.

Now, regarding the group, not sure if failure to create a group could be the root cause. Guess we need to have a look at that.

alphahotel29, 2020-02-10:
Yes, sorry for that. You're right, pairing is functioning. And I can read buttonevents. So pairing is good. Regarding the group, something fail. I've reproduce this after multiple reset of the gateway, and in different order. Seem this ikea light /remote need a group to work together (when gateway is down).  Also, I've seen that I'm not on the latest version of phoscon app. (2.5.71). I will try with the latest one.  

Do I need to open a new issue for that? 

Thanks !

SwoopX, 2020-02-11:
I'd recommend opening a new issue for this. Btw, have you tried setting up a corresponsing group manually?

hmrc87, 2020-02-11:
thanks for fixing this issue @SwoopX, unfortunately I could not test it yet, since I cannot update my phoscon gateway. I'm still looking into the other github issues to figure out how I can update it. I will report asap if the fix worked (i really think it will)

jan666, 2020-02-21:
I habe the same problem.

I have 4 Tradfi Remotes. One is on firmware 1.2.223, the 3 others are on firmware version 2.3.014. 

The 1.2.223 one works perfect, I can access the group on the left side in the phoscon app, assign it to lights an everything is perfect.

The three on 2.3.014 create groups on the left side in the phoscon app, but I cant enter them to assign it to lights. 

Console logs (Chrome):

click on 1.2.223:

```
current site: group/lights
main.js?8603a4:3 goto: control/switch/<id>/group/**37606**
```

click on one of the 2.3.014: 

`goto: control/switch/<id>/group/**undefined**`

SwoopX, 2020-02-21:
Hm, looks like the device behaves differently with the new firmware. Could you guys close this issue then an open another one for the group thingy, so we stay clean with the topics?

Happy to have a look then but I have very limited experience with lights as I barely use them.

jan666, 2020-02-23:
> Hm, looks like the device behaves differently with the new firmware. Could you guys close this issue then an open another one for the group thingy, so we stay clean with the topics?
> 
> Happy to have a look then but I have very limited experience with lights as I barely use them.

I added  #2485

robidog, 2020-04-19:
Has anyone sorted out the Tradfri Remote Control pairing issue in the meantime. Stuck at the 10 sec button press phase as well. Remote is visible and responsive in deCONZ GUI, but not usable in Phoscon. FW of remote is 1.2.223, deCONZ 2.05.75, RaspBee 0x26050500.

anduchs, 2020-04-22:
I updated the RaspBee Firmware to 0x26**3**50500 (note the 3rd diget) and then (newer) Ikea worked for me.
However Updating via Homeassistant-Deconz-WebGUI did not work. I had to get a raspbian on my Pi and update via command line as described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

hmrc87, 2020-05-09:
After finally managing to update my firmware to 26050500 (RaspBee) I was able to pair my Ikea Tradfri Remote.

Thank you so much and sorry for the late reply

closed 2020-05-09 by hmrc87

robidog, 2020-05-10:
Can confirm that 0x26350500 solves the issue with Tradfri Remote. As mentioned in this thread by @anduchs and [here](https://github.com/marthoc/docker-deconz/issues/187#issuecomment-626153490), I had to remove the RaspBee from my Pi4 running HassOS and do the update on an older Pi running Raspbian.