dresden-elektronik/deconz-rest-plugin #2247: Use "not yet supported" devices

chrisrock1984, 2019-12-26:
dear all,

I bought a smart9 zigbee alarm remote -> https://www.aliexpress.com/i/4000179913087.html?spm=2114.12057483.0.0.38d55277rXMzxO

I was able to connect it with deconz - at least it was visible in the VNC environment. My question is, if it is possible to use such a device, even if it is still not supported within node-red and some filtering?
I was not able to find a common output node, which outputs all zigbee messages within the network.

A "template" feature like the one in the tasmota firmware would be a nice feature to add new devices much easier!

Thank you all for your help!

Smanar, 2019-12-26:
If you can see it in the GUI and if it's stable it's a good starting point.
You can take a look in the api to check what is missing.
But for missing thing you need to edit the c++ code, you have the procedure on the homepage of the github.
If you are lucky it can works just with adding it in whitelist, look at the old pull request for a device that is like yours.

And yes, I hope for a easier system to add new devices in the future API.

chrisrock1984, 2019-12-26:
Thank you for your answer.
I am using the hassio addon of deconz. Is there a possibilitiy to add it to the whitelist with the docker version of deconz?

chrisrock1984, 2019-12-26:
The device is only visible in the VNC desktop as node but not accessible via the api. It was not in the list, when I did the GET config

Smanar, 2019-12-26:
Ha yep, it's normal.
You need to include it using phoscon to use the API. If you use only deconz, the device will not add in the API.

chrisrock1984, 2019-12-26:
I added the device in phoscon but it never showed up there. At the same time I had the VNC deconz GUI open and saw, that a new device appeared. How can I use that device?

Smanar, 2019-12-27:
It's normal.
You need to take a look in the API to see it. your home automation application can read it too (but IDK how work node-red for deconz).

And if you haven't the device in the API, you need to add it in the deconz api code.

accelle17, 2020-01-05:
already have an open request device support since last year for similar device but no response yet - #1707 

It seems the only way is to learn to edit the c++ code and the api or to switch to zigbee2mqtt which already supports the device.

closed 2020-05-13 by stale[bot]