dresden-elektronik/deconz-rest-plugin #2248: Device Support Request Bitron 902010/32

easybeat, 2019-12-27:
Hi

It would be really good if this device would be supported. Here the link [https://www.amazon.de/Bitron-902010-32-BITRONVIDEO-Wandtermostat-ZigBee/dp/B016V7IJRG](url)

I think a device like that is missing to integrate a heating system.

Kind regards

Beat

rgraenitz, 2019-12-29:
I actually searching for a compatible wall thermostat so that not just the light can be controlled. The only Zigbee thermostat I found was the above one .... did anybody already check if its compatible with deCONZ??

SwoopX, 2020-01-02:
Sould already be supported according to #1003

easybeat, 2020-01-04:
Thank you for this information! I will order one and test.

Coalado, 2020-01-07:
@easybeat  Please give feedback

closed 2020-05-13 by stale[bot]

easybeat, 2020-06-29:
@Coalado I finally got it tested and the device is recognized in Deconz but does not show up in Phoscon app.

And I'm using Deconz adapter in ioBroker and there I can see it and it also updates the current temperature BUT if I change the setpoint on the device, this value is not coming through to ioBroker. But if I change the setpoint in ioBroker it shows up on the device.

Can I test if the value is shown in Deconz if I change it on the device? This would mean that there is a problem with the ioBroker adapter and this here could be closed.

Thank you for help.

Beat

SwoopX, 2020-06-29:
> Can I test if the value is shown in Deconz if I change it on the device? This would mean that there is a problem with the ioBroker adapter and this here could be closed.

You can easily do that via deconz GUI, if you read the attributes of the corresponding cluster (I presume thermostat cluster).

easybeat, 2020-09-21:
> > Can I test if the value is shown in Deconz if I change it on the device? This would mean that there is a problem with the ioBroker adapter and this here could be closed.
> 
> You can easily do that via deconz GUI, if you read the attributes of the corresponding cluster (I presume thermostat cluster).

Hi

my current deconz version is 2.5.82.

Ok I finally managed to do the test and investigate the problem. The attribute is fine in deconz:
[image]

Not sure if it is working ok as I had to press "read" button all the time after I changed the setpoint on the device? Should it not update automatically?

Because what I figured out is that if I press the read button the value get update also in ioBroker where I use the deconz adpater:
[image]

So my question is where is the problem? Should the setpoint automatically update in deconz after I change it on the device or does the ioBroker adapter have to send a "read" command to the attribute all the time?

Thank you for your help.

Kind regards

Beat

SwoopX, 2020-09-21:
Looks like the attribute reporting is not defined. You may want to try this https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting

easybeat, 2020-09-22:
OK will try this but is this a bug on this device integration?

Links:
------
[1] 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2248#issuecomment-696355479
[2] 
https://github.com/

SwoopX, 2020-09-22:
No, it's not specific to the device. More a topic for the default settings.

easybeat, 2020-09-25:
Hi

yes after I configured the attribute reporting it is now working!

Thank you very much.

Kind regards

Beat

Links:
------
[1] 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2248#issuecomment-696355479
[2] 
https://github.com/

easybeat, 2021-02-02:
@SwoopX Hi, I just realised that the battery level is also not "reported" and I don't get it in ioBroker at all. Is this a similar problem or could that be not handled correctly in the ioBroker adaptor?

[image]
[image]

Mimiix, 2021-02-02:
I dont see the battery percentage in there at all. Just the voltage ? The cluster for that is greyed out afaik.