dresden-elektronik/deconz-rest-plugin #2249: Trouble with Conbee II after couple of weeks

connoleg, 2019-12-27:
Hi,
Recently installed Conbee II into my HA Intel NUC based HASS.IO system. Immediately able to add a couple hue lights via the Deconz plugin. All working OK for a few weeks then the two lights became unresponsive. I have tried various things suggested on GitHub and the HA forums, reinstalled, rescanned but no lights found. I moved Conbee II to another test system located about 15" from a light and scanned, still finds no lights. Everything looks OK (to me) in the Deconz log:
```
12:06:46:606 Announced to internet https://phoscon.de/discover
12:07:45:310 Current channel 15
12:07:45:318 Device TTL 1558 s flags: 0x7
12:08:45:311 Current channel 15
12:08:45:319 Device TTL 1498 s flags: 0x7
12:09:45:309 Current channel 15
12:09:45:315 Device TTL 1438 s flags: 0x7 
```
Here are my versions etc:
- ConBee II
- Version 2.05.72
- Firmware 264A0700
- Deconz Hass.io V5

Any ideas about further fault finding? Is the Conbee II potentially faulty
Thank you

klwlau, 2019-12-28:
I got similar problem too. Conbee II can read state from sensors but cannot send command to device, such as turn on or off lights/switches.

PremiumUsername, 2019-12-30:
I lose lights with that version too. Try 2.05.69. When I downgraded I still had to power cycle the lamps I lost  to get them reconnected but after that it has been as stable as can be.

closed 2020-05-06 by stale[bot]