dresden-elektronik/deconz-rest-plugin #2250: Aqara devices not handled

robUx4, 2019-12-28:
Following [this issue](https://github.com/zigpy/zigpy-deconz/issues/92) I'm trying to find a solution to pair my Aqara devices with the Conbee II in Home Assistant.

Basically the devices (2 models of temperature sensors, a motion sensor and a dual button wall switch) are never recognized by deconz.

One motion sensor looks like this in VNC after trying to add it:
[image]

In the logs I get something like this:
```
09:21:11:076 Mgmt_Lqi_req zdpSeq: 2 to 0x00212EFFFF0481F2 start index 0
09:21:11:076 APS-DATA.request id: 133, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:21:11:092 APS-DATA.confirm id: 133, status: 0x00 SUCCESS
09:21:11:093 APS-DATA.confirm request id: 133 -> confirmed, timeout 24407024
09:21:11:099 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:21:11:099 APS-DATA.indication request id: 133 -> finished
09:21:11:099 APS-DATA.request id: 133 erase from queue
09:21:11:099 ZDP status = 0x00 -> SUCCESS
09:21:11:099 ZDP Mgmt_Lqi_rsp zdpSeq: 2 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:21:11:099     * neighbor: 0x00158D000353615D (0x298D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
09:21:13:956 Mgmt_Lqi_req zdpSeq: 12 to 0x00212EFFFF0481F2 start index 0
09:21:13:956 APS-DATA.request id: 144, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:21:13:971 APS-DATA.confirm id: 144, status: 0x00 SUCCESS
09:21:13:971 APS-DATA.confirm request id: 144 -> confirmed, timeout 24046632
09:21:13:976 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:21:13:976 APS-DATA.indication request id: 144 -> finished
09:21:13:976 APS-DATA.request id: 144 erase from queue
09:21:13:976 ZDP status = 0x00 -> SUCCESS
09:21:13:976 ZDP Mgmt_Lqi_rsp zdpSeq: 12 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:21:13:976     * neighbor: 0x00158D000353615D (0x298D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
09:21:16:835 Mgmt_Lqi_req zdpSeq: 22 to 0x00212EFFFF0481F2 start index 0
09:21:16:836 APS-DATA.request id: 155, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:21:16:851 APS-DATA.confirm id: 155, status: 0x00 SUCCESS
09:21:16:851 APS-DATA.confirm request id: 155 -> confirmed, timeout 21669008
09:21:16:856 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:21:16:856 APS-DATA.indication request id: 155 -> finished
09:21:16:856 APS-DATA.request id: 155 erase from queue
09:21:16:856 ZDP status = 0x00 -> SUCCESS
09:21:16:856 ZDP Mgmt_Lqi_rsp zdpSeq: 22 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:21:16:856     * neighbor: 0x00158D000353615D (0x298D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
09:21:17:735 CTRL skip polling while permit join is set
09:21:17:956 CTRL skip polling while permit join is set
09:21:18:155 CTRL skip polling while permit join is set
09:21:18:354 CTRL skip polling while permit join is set
09:21:18:552 CTRL skip polling while permit join is set
09:21:18:751 CTRL skip polling while permit join is set
09:21:18:949 CTRL skip polling while permit join is set
09:21:19:149 CTRL skip polling while permit join is set
09:21:19:347 CTRL skip polling while permit join is set
09:21:19:545 CTRL skip polling while permit join is set
09:21:19:716 Mgmt_Lqi_req zdpSeq: 32 to 0x00212EFFFF0481F2 start index 0
09:21:19:716 APS-DATA.request id: 166, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:21:19:732 APS-DATA.confirm id: 166, status: 0x00 SUCCESS
09:21:19:732 APS-DATA.confirm request id: 166 -> confirmed, timeout 24061672
09:21:19:737 APS-DATA172.30.32.2 - - [28/Dec/2019:09:21:19 +0100] "GET /api/8047A5ACEB/sensors/new?_=1577521252722 HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:20 +0100] "GET /api/8047A5ACEB/sensors?_=1577521252723 HTTP/1.1" 200 436 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:20 +0100] "GET /api/8047A5ACEB/sensors/new?_=1577521252724 HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:20 +0100] "GET /api/8047A5ACEB/sensors/new?_=1577521252725 HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:21 +0100] "PUT /api/8047A5ACEB/config HTTP/1.1" 200 54 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:21 +0100] "GET /api/8047A5ACEB/sensors/new?_=1577521252726 HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-sensors.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:21:23 +0100] "GET /websocket HTTP/1.1" 101 4 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
09:21:19:737 APS-DATA.indication request id: 166 -> finished
09:21:19:738 APS-DATA.request id: 166 erase from queue
09:21:19:738 ZDP status = 0x00 -> SUCCESS
09:21:19:738 ZDP Mgmt_Lqi_rsp zdpSeq: 32 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:21:19:738     * neighbor: 0x00158D000353615D (0x298D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
09:21:22:595 Mgmt_Lqi_req zdpSeq: 42 to 0x00212EFFFF0481F2 start index 0
09:21:22:595 APS-DATA.request id: 177, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:21:22:608 APS-DATA.confirm id: 177, status: 0x00 SUCCESS
09:21:22:608 APS-DATA.confirm request id: 177 -> confirmed, timeout 24046632
09:21:22:613 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:21:22:613 APS-DATA.indication request id: 177 -> finished
09:21:22:613 APS-DATA.request id: 177 erase from queue
09:21:22:613 ZDP status = 0x00 -> SUCCESS
09:21:22:613 ZDP Mgmt_Lqi_rsp zdpSeq: 42 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:21:22:613     * neighbor: 0x00158D000353615D (0x298D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
```

It seems none of my devices are recognize with their Zigbee profile. They are working fine with the Aqara Gateway, one press during discovery and they are added.

The ConBee II uses the latest firmware (2.05.72 12/12/2019 / 26490700).
I'm using a fresh install of Home Assistant 0.103.4 on Raspberry Pi 3B+

robUx4, 2019-12-28:
In addition to the devices mentioned in the other posts, here are the logs for an Aqara ZNLDP12LM light:

```
172.30.32.2 - - [28/Dec/2019:09:43:44 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:44 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:45 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:45 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:46 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:47 +0100] "PUT /api/8047A5ACEB/config HTTP/1.1" 200 54 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:47 +0100] "GET /api/8047A5ACEB/sensors/new HTTP/1.1" 200 36 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:48 +0100] "GET /api/8047A5ACEB/lights?_=1577522592967 HTTP/1.1" 200 12 "http://hassio.local:8123/api/hassio_ingress/xFphmbccq1liG6x98-Z11MBb2rPjY63yofKyp0UDY-I/pwa/devices-lights.html?_v=bb47cdbc19c257e3000fdb8475aa456830e6df94,11,6,9&gwid=00212EFFFF0481F2" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
172.30.32.2 - - [28/Dec/2019:09:43:48 +0100] "GET /websocket HTTP/1.1" 101 4 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:71.0) Gecko/20100101 Firefox/71.0"
09:43:33:966 APS-DATA.request id: 137 erase from queue
09:43:33:966 ZDP status = 0x00 -> SUCCESS
09:43:33:966 ZDP Mgmt_Lqi_rsp zdpSeq: 51 from 0x00212EFFFF0481F2 total: 0, startIndex: 0, listCount: 0
09:43:36:827 Mgmt_Lqi_req zdpSeq: 61 to 0x00212EFFFF0481F2 start index 0
09:43:36:827 APS-DATA.request id: 148, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:43:36:843 APS-DATA.confirm id: 148, status: 0x00 SUCCESS
09:43:36:843 APS-DATA.confirm request id: 148 -> confirmed, timeout 24230872
09:43:36:849 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:43:36:849 APS-DATA.indication request id: 148 -> finished
09:43:36:849 APS-DATA.request id: 148 erase from queue
09:43:36:849 ZDP status = 0x00 -> SUCCESS
09:43:36:849 ZDP Mgmt_Lqi_rsp zdpSeq: 61 from 0x00212EFFFF0481F2 total: 1, startIndex: 0, listCount: 1
09:43:36:849     * neighbor: 0x00158D0002D71C56 (0xA2F7), LQI: 225, relation: 0x01 rxOnWHenIdle: 1
09:43:37:313 new node - ext: 0x00158d0002d71c56, nwk: 0xA2F7
09:43:39:708 Mgmt_Lqi_req zdpSeq: 69 to 0x00158D0002D71C56 start index 0
09:43:39:708 APS-DATA.request id: 157, addrmode: 0x03, addr: 0x00158d0002d71c56, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:43:45:981 dev /dev/ttyAMA0
09:43:45:982 GW firmware version: 0x26490700
09:43:45:982 GW firmware version is up to date: 0x26490700
09:43:46:286 CTRL skip polling while permit join is set
09:43:46:943 saved node state in 1 ms
09:43:47:963 sync() in 1020 ms
09:43:48:872 Websocket disconnected 127.0.0.1:50832 (state: 0) 
09:43:49:842 APS-DATA.confirm id: 157, status: 0xD0 
09:43:50:300 Mgmt_Lqi_req zdpSeq: 94 to 0x00212EFFFF0481F2 start index 0
09:43:50:300 APS-DATA.request id: 182, addrmode: 0x03, addr: 0x00212effff0481f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
09:43:50:314 APS-DATA.confirm id: 182, status: 0x00 SUCCESS
09:43:50:314 APS-DATA.confirm request id: 182 -> confirmed, timeout 24063192
09:43:50:320 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:43:50:320 APS-DATA.indication request id: 182 -> finished
[... cut ...]
09:43:57:033 APS-DATA.confirm id: 203, status: 0x00 SUCCESS
09:43:57:033 APS-DATA.confirm request id: 203 -> confirmed, timeout 24091808
09:43:57:039 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
09:43:57:039 APS-DATA.indication request id: 203 -> finished
09:43:57:039 APS-DATA.request id: 203 erase from queue
```

At this point I'm wondering if my  Conbee II is defective. It never managed to detect any of my devices properly.

ebaauw, 2019-12-28:
> One motion sensor looks like this in VNC after trying to add it

Looks like deCONZ hasn’t read the _Simple Descriptors_.  I’ve seen some intermittent issue that deCONZ sometimes doesn’t want to read these while the network is open.  Not sure what’s causing this, but I don’t think it’s hardware related.

Make sure the network is closed (search has finished), then read the descriptors from the GUI (from the drop down menu behind the left circle on the node).  Make sure you wake the sensor, by waving in front of it or briefly pressing the button.  After the descriptors have been read, the right circle should appear, as should the dropdown behind it, with the endpoints and clusters.

To continue the pairing process, search for devices from Phoscon or the old web app.  In the _Cluster Info_ panel read the attributes of the _Basic_ cluster.  Again, make sure the sensor is awake.  That should trigger the creation of the REST resources, and the label on the node should change to the value of the `name` attribute in the resource, from the NWK address.

robUx4, 2019-12-28:
Thanks for the tip, I didn't know it was possible to use these "circles". I only have one on the right of each device, whereas the Coordinator has two with a "F2 ZLL : Green Power / GP Commission Tool". Is that expected ?

When I select the "(000) Read Simple Descriptor(s)" of each device the square stays pressed for 1s and goes away. But I don't see the second circle. I tried pressing the button on the device before or after I read the descriptor, but nothing changes.

[image]

robUx4, 2019-12-28:
This seems like a similar issue to #2043

I tried deCONZ 2.05.71 and 2.05.69 on Windows. Now I see the Level Switch in addition to the Home Automation and Green Power clusters (?). But still not way to read the simple descriptor of my devices.

Adminiuga, 2019-12-28:
> When I select the "(000) Read Simple Descriptor(s)" of each device the square stays pressed for 1s 

When trying to read anything from aqara devices, you need to wake them up  by pressing the button. So request the simple descriptor, then press the button on the device a few times with a 1.5s interval.

robUx4, 2019-12-29:
I did that. I think the problem here is that the "Read Simple Descriptor(s)" button doesn't stay on at all. I assume that's either because it read instantly the data (unlikely) or that it's failing for some reason (are there logs somewhere ?).

Also odd thing in that software. No matter how many times I delete a device, it reappears almost instantly.

closed 2020-05-06 by stale[bot]

robUx4, 2021-03-01:
For the record, I reset my ConBee II through the Phoscon App and now it can see Aqara and Lidl device correctly in Home Assistant.