dresden-elektronik/deconz-rest-plugin #2251: Reporting for Consumption.

Smanar, 2019-12-28:
Hello, I m boring with all reporting I can see in logs for Consumption/voltage/current, every X minuts always unchanged and 0 value.

Home applications realy need them ?

If I block reporting for 0 value and unchanged, a device can be marked as unavailable after some hours I think ?

ebaauw, 2019-12-28:
What do you mean by reporting? The deCONZ log or the web socket notifications?  In case of the log, simply run deCONZ with fewer debug options.  In case of web socket, set `/config/websocketnotifyall` to false to get notifications only for changed values.  You’ll still get the changed `state.lastupdated`, though.

Smanar, 2019-12-28:
Web socket notifications, and I can't change the websocketnotifyall mode else the plugin I m using will stop working.

It's realy something usefull ? All network are powerfull enought to manage that, but for me it's "unnecessary chatter"

ebaauw, 2019-12-28:
> I can't change the websocketnotifyall mode else the plugin I m using will stop working.

That's probably a bug with the plugin you're using.  It should work with both settings; homebridge-hue does.

I do feel the `notifyall` would better have been a setting per web socket client (channel) instead of a single global setting.  However that would require the REST API plugin also to accept input over the web socket, with all security implications.

> It's realy something usefull ?

Yes.  It allows an application to see when there's been communication from the device.

> for me it's "unnecessary chatter"

In that case, change the attribute reporting settings of your smart plugs in the deCONZ GUI.  Don't mess with the events in the REST API plugin; the rule execution also depends on this.

Smanar, 2019-12-28:
In my situation I have take a look at the home application.

If I have a 50W consumption at 10h, and still 50W during 1 hour, I just need a second notification at 11h with 0W, the application keep the same value as long as it haven't new value.
I realy don't need a notification every 10s if the value is the same (especially if this one is zero)

But if I set the attribute reporting every minuts, it's impossible having something reliable, the consumption can move a lot in 60s.

For me, as it, the Consumption need to be used with websocketnotifyall to false, but if you prefer set it to true for others devices ...

In fact I just don't see the utility using Consumption feature with this notification spamming.

I know it same for exemple for xiaomi sensor, that send useless informations every hours even nothing have changed, but not comparable with Consumption notification you can have, if you have 10 plugs.

SwoopX, 2020-03-01:
@Smanar Hey, I also recently investigated my zigbee network traffic to see if I can get the air "more quite". I'm running my home automation (deconz and FHEM) on a pi zero and although I have plenty devices meanwhile, there's still reserves for more. However, I'm always looking for chances to further optimize performance and unnecessary chatter.

One thing I recently observed (and which disturbs me), was that my smart plugs obviously created tons of (unnecessary) events. In consequence, the database where I log my power and consumption values grew way bigger than expected. To cease that, I firstly created filters in FHEM to let only power and consumption changes through, but that is unwanted processing power usage.
Next, I took an eye on the attribute reporting and disabled it for current and voltage, never really used them though. That was an observerable improvement. However, I could still see that a value refresh in FHEM stayed nearly the same. Very strage.
Last week, I took another close view on the zigbee traffic. "Noise" had indeed decreased, but I still saw polling of certain attributes for my innr SP120, consumption and instantaneous demand. That puzzled me since consumption was reported upon (settings 1, 300, 1) as well. The polling occurs, as I understand it, because instantaneous demand is always polled on the electrical measurement cluster, even for devices where it's not available.

Now to the point: one of my last commits should finally prevent polling if instantaneous demand attribute is not available. Not sure if the consumption device you're referring to supports it hepl in you case as well. If you then tweak you attribute reporting according to your needs, it should become really quiet. I mean, if you have a really steady value (e.g. if the smart plug or whatsoever is off), than you could increase the mx reporting interval to an hour or to 65534.

For my use case, I overlooked the last piece of the puzzle with the comment from @ebaauw on the websocket notifications. I coincidentially observed that those still also contain current and voltage where I've disabled attribute reporting for and that's why I also see those values being updated in FHEM.

So bottom line is changing websocket notification configuration should help to get rid of the additional filtering in FHEM freeing more resources me being a happier man ;)

ebaauw, 2020-03-02:
Note that in the reporting config, you set three values: min, max, and change.

Min indicates how frequently changes are reported, if you set it to 1, expect up to one change report every second.  If you set it to 60, you’ll receive one change report per minute max.

Max indicates how often period reports are sent.  You want to use these for heartbeats and for slowly changing attributes, like battery and maybe total consumption.

Change sets the threshold for sending a change report.  Because the different devices use different scaling, this takes some double-checking.  Also, I find the measurements of most devices aren’t very accurate, resulting in a lot of small fluctuations.  E.g. if your voltage measurements routinely fluctuate between 223V and 233V, you want to set change to a value larger than 10V. 

You can enable or disable change reports and periodic reports independently.  I think you need to
set min to 65535 to disable reports on change; set max to 65535 to disable periodic reports.

SwoopX, 2020-03-02:
@ebaauw thanks for adding the details around reporting. Spared those out, but very important as well.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix