dresden-elektronik/deconz-rest-plugin #2252: Alexa cant discover my deviceI(s)

Craig01988, 2019-12-28:
I allow access to the device and get the 60 second timer, ask alexa to discover devices but nothing. Any suiggestions?

kmplngj, 2019-12-29:
>I allow access to the device and get the 60 second timer, ask alexa to discover devices but nothing. Any suiggestions?

@Craig01988 Tell a bit more about your setup. How do you run Deconz?

Craig01988, 2019-12-29:
Hi, I'm running deconz via hassio on a rpi4, I toought I may be able to link with out relying on the cloud but I've enabled the ha cloud service trial and can link to Alexa that way 😁. Is it possible without cloud services?

kmplngj, 2019-12-29:
I think the Hassio Deconz Add On will run on another port than port 80. 
Deconz need to be running on Port 80 to work with Alexa.

Perhaps you could ask @frenck if this is so.

closed 2020-05-06 by stale[bot]