dresden-elektronik/deconz-rest-plugin #2253: Deconz gui, cannot reach it

Max0212, 2019-12-28:
I have installed deconz on my RPi4 with the latest raspian Buster. 

Version 2.05.71
Firmware 264A0700

I have used the image file: 

```
deCONZ 2.05.67 Version 9-2019 released 8-9-2019
Download Stable
md5: a6901560340722bf02757a63c665ad14
```
And everything is running fine, my devices are connected and working.

The thing I have not been able to do is to connect to the deconz gui using realVNC. The deconz-gui is active (see sample below). When I click the icon on the VNC viewer desktop, nothing happens.
I think it is user/permission related but I cannot figure it out.

I did:

```
sudo systemctl disable deconz
sudo systemctl stop deconz
```

There are several issues covering this subject but non of them gives (me ) a clue.

I hope someone can help me, I need this to connect my Eurotronic spirit zigbee valves.

Thanks in advance

```
pi@phoscon:~ $ systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor prese
   Active: active (running) since Tue 2019-12-24 16:22:18 CET; 4 days ago
 Main PID: 1078 (deCONZ)
    Tasks: 8 (limit: 4915)
   Memory: 39.0M
   CGroup: /system.slice/deconz-gui.service
           ├─1078 /usr/bin/deCONZ --http-port=80
           ├─1096 dbus-launch --autolaunch 542207c800484554be0d0a6a2f0467a0 --bi
           └─1097 /usr/bin/dbus-daemon --syslog-only --fork --print-pid 5 --prin

dec 24 16:22:18 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST A
dec 24 16:22:19 phoscon deCONZ[1078]: QStandardPaths: XDG_RUNTIME_DIR not set, d
dec 24 16:22:19 phoscon deCONZ[1078]: libpng warning: iCCP: known incorrect sRGB
dec 27 17:40:46 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: U
dec 27 17:41:18 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: U
dec 27 17:50:42 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: U
dec 27 17:51:28 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: U
lines 1-18/18 (END)...skipping...
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-12-24 16:22:18 CET; 4 days ago
 Main PID: 1078 (deCONZ)
    Tasks: 8 (limit: 4915)
   Memory: 39.0M
   CGroup: /system.slice/deconz-gui.service
           ├─1078 /usr/bin/deCONZ --http-port=80
           ├─1096 dbus-launch --autolaunch 542207c800484554be0d0a6a2f0467a0 --binary-syntax --close-stderr
           └─1097 /usr/bin/dbus-daemon --syslog-only --fork --print-pid 5 --print-address 7 --session

dec 24 16:22:18 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
dec 24 16:22:19 phoscon deCONZ[1078]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
dec 24 16:22:19 phoscon deCONZ[1078]: libpng warning: iCCP: known incorrect sRGB profile
dec 27 17:40:46 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
dec 27 17:41:18 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
dec 27 17:50:42 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
dec 27 17:51:28 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
```
```

Smanar, 2019-12-28:
Can you try with turn them off both

```
sudo systemctl stop deconz-gui
sudo systemctl stop deconz
```

And after launch the application by the menu ?

Max0212, 2020-01-05:
Thanks for your response, I've got it working now. Not sure what happened

Acer54, 2020-01-06:
For others with the same problem, I have made a step-by-step guide [here](https://my-darling-is-a-linux-user.org/?p=827).
Included is a JS-Script, how to set the heatpoint using the REST-API.
... The whole description is in German Language... so you might need to use a translator.

closed 2020-05-13 by stale[bot]