dresden-elektronik/deconz-rest-plugin #2255: Can't pair new IKEA Tradfri GU10 Bulbs

WebOscar, 2019-12-29:
I have a really new ConBee II setup on a Raspi with the stable Raspbian Buster Desktop image from Dresden Elektronik.

After fresh install and updates:
Version:  2.05.71 / 17.11.19
Firmware: 264A0700

I was able to pair a lot of IKEA Tradfri Bulbs (E27, E14 & GU10), IKEA Tradfri PIR's and Tradfri Switches. Actually I have 40 devices connected.

Yesterday I bought three new Tradfri GU10 white bulbs. They differ from the old ones, by looking like real glass-bulbs, while the older ones are out of white plastic.

Any Ideas, why I can't pair the new GU10-Bulbs?

WebOscar, 2019-12-29:
I'm sorry! This issue can be closed.
Solution: After hours of trying everything, including downgrading deCONZ, I found a post, which said "switch on and off six times". I always switched eight times, as this worked for the old bulbs. Even the new bulbs starts to flicker, when I used the eight-time-switcher. But as I tried with six times, the pairing worked immediately.

closed 2019-12-29 by WebOscar

antonbo, 2020-02-16:
Thank for the hint. It works for me as well. One small addition: light should be on for at least 700-1000 ms.