dresden-elektronik/deconz-rest-plugin #2256: "Random" irresponsive lights (tradfri)

PeterRemnemark, 2019-12-29:
Have an issue that has been for a while and identified after I updated my install with several new lights to my zigbee network.

I've Conbee 1, Deconz in docker container (running at synology) version 2.05.17 and fw 26330500.
I've changed the WiFi channel to opposite frequency (ch 1) and set Conbee for ch 25. No other neighbours are in the same spectrum (either 1 or 6 with very low signal and far distance). Have put the stick on a extension cord for a better reception.

With random, I refer to a group of lights that within that group random stops to react to commands by either automations (home-assistant) or via Phoscon. It is not the same type of Tradfri model, it is different and both new and old ones that has been for a long period within the network (and worked for that time).

Below is the log that I recently captured when I discovered one faulty light. The light that did not work is the **0x14B457FFFE7B1011.** For simplicity, I've cut out some repeating lines, but kept the number of them that repeated if that is needed. Let me know what I can do for identify the issue

16:52:08:222 Current channel 25
,16:52:08:248 Device TTL 2580 s flags: 0x7
,16:52:10:466 Announced to internet
,16:52:20:164 remove outdated neighbor 0x3735
,16:52:22:063 sensor 2 (lumi.sensor_motion): disable presence
,16:52:25:162 remove outdated neighbor 0xA265
,16:52:26:318 reuse dead link (dead link container size now 22)
,16:52:28:825 no button map for: lumi.sensor_magnet ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
,16:52:28:825 ZCL attribute report 0x00158D00019DB522 for cluster 0x0006, ep 0x01
,16:52:35:162 remove outdated neighbor 0xA265
,16:52:35:662 binding for cluster 0x0000 of 0x14B457FFFE7B2287 exists (verified by reporting)
,16:52:35:663 binding for cluster 0x0006 of 0x14B457FFFE7B2287 exists (verified by reporting)
,16:52:35:663 binding for cluster 0x0008 of 0x14B457FFFE7B2287 exists (verified by reporting)
,16:52:36:042 ZCL configure reporting rsp seq: 37 0x14B457FFFE7B2287 for cluster 0x0008 attr 0x0000 status 0x00
,16:52:39:045 ZCL configure reporting rsp seq: 36 0x14B457FFFE7B2287 for cluster 0x0006 attr 0x0000 status 0x00
,16:52:43:297 ZCL attribute report 0x000B57FFFE89891B for cluster 0x0000, ep 0x01
,16:52:45:521 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
,16:52:45:521 ZCL attribute report 0x00158D0002E98F6D for cluster 0x0402, ep 0x01
,16:52:45:540 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
,16:52:45:540 ZCL attribute report 0x00158D0002E98F6D for cluster 0x0405, ep 0x01
,16:52:45:564 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
,16:52:45:564 ZCL attribute report 0x00158D0002E98F6D for cluster 0x0403, ep 0x01
,16:52:46:238 no button map for: lumi.sensor_motion ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
,16:52:46:238 ZCL attribute report 0x00158D0001B15D50 for cluster 0x0406, ep 0x01
,16:52:50:162 remove outdated neighbor 0x16F2
,16:52:50:162 remove outdated neighbor 0x9D11
,16:52:50:162 remove outdated neighbor 0x87E7
,16:52:55:162 remove outdated neighbor 0x0AA4
,16:53:05:453 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
,16:53:05:453 ZCL attribute report 0x00158D0001D36CA4 for cluster 0x0402, ep 0x01
,16:53:05:493 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
,16:53:05:493 ZCL attribute report 0x00158D0001D36CA4 for cluster 0x0405, ep 0x01
,16:53:05:517 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
,16:53:05:517 ZCL attribute report 0x00158D0001D36CA4 for cluster 0x0403, ep 0x01
,16:53:06:062 sensor 10 (lumi.sensor_motion): disable presence
,16:53:06:410 	0x14B457FFFE72B1E0 force poll (2)
,16:53:06:524 0x14B457FFFE72B1E0 level 1 --> 187
,16:53:06:524 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:07:524 0x14B457FFFE72B1E0 level 187 --> 1
,16:53:07:525 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:07:846 	0x14B457FFFE72B1E0 force poll (2) 
,16:53:08:101 	0x14B457FFFE72B1E0 force poll (2)
,16:53:08:216 Current channel 25
,16:53:08:242 Device TTL 2521 s flags: 0x7
,16:53:08:530 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:09:503 	0x14B457FFFE72B1E0 force poll (2)
,16:53:09:610 0x14B457FFFE72B1E0 level 255 --> 68
,16:53:09:610 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:10:162 remove outdated neighbor 0xC6F8
,16:53:10:162 remove outdated neighbor 0xAF01
,16:53:10:162 remove outdated neighbor 0x1615
,16:53:10:614 0x14B457FFFE72B1E0 level 68 --> 254
,16:53:10:614 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:12:996 	0x14B457FFFE72B1E0 force poll (2)
,16:53:13:101 0x14B457FFFE72B1E0 level 1 --> 188
,16:53:13:101 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:14:132 0x14B457FFFE72B1E0 level 188 --> 1
,16:53:14:133 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
,16:53:14:464 	0x14B457FFFE72B1E0 force poll (2)
,16:53:14:618 	0x14B457FFFE72B1E0 force poll (2)
,16:53:15:102 ZCL attribute report 0x14B457FFFE72B1E0 for cluster 0x0008, ep 0x01
[... 684 lines cut ...]
,16:56:17:836 ZCL configure reporting rsp seq: 101 0x000B57FFFE89891B for cluster 0x0006 attr 0x0000 status 0x00
,16:56:17:886 ZCL configure reporting rsp seq: 102 0x000B57FFFE89891B for cluster 0x0008 attr 0x0000 status 0x00
,16:56:18:043 verified group capacity: 255 and group count: 2 of LightNode 0x000b57fffe89891b
,16:56:18:043 0x000b57fffe89891b found group 0x0001
,16:56:18:043 0x000b57fffe89891b found group 0xFFF0
,16:56:42:530 ZCL attribute report 0x000B57FFFED831EC for cluster 0x0000, ep 0x01
,16:57:08:214 Current channel 25
,16:57:08:240 Device TTL 2281 s flags: 0x7
,16:57:31:287 no button map for: lumi.sensor_motion ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
,16:57:31:287 ZCL attribute report 0x00158D0001231FFA for cluster 0x0406, ep 0x01
,16:57:32:162 Set sensor check interval to 100 milliseconds

Filtered out the last lines which mention the keyword "0x14B457FFFE7B1011"
-------
16:54:29:175 max transmit errors for node 0x14B457FFFE7B1011, last seen by neighbors 53 s
17:07:59:854 ZCL attribute report 0x14B457FFFE7B1011 for cluster 0x0000, ep 0x01
17:07:59:879 ZCL attribute report 0x14B457FFFE7B1011 for cluster 0x0000, ep 0x01
17:07:59:981 ZCL attribute report 0x14B457FFFE7B1011 for cluster 0x0000, ep 0x01
17:12:27:663 binding for cluster 0x0000 of 0x14B457FFFE7B1011 exists (verified by reporting)
17:12:27:664 binding for cluster 0x0006 of 0x14B457FFFE7B1011 exists (verified by reporting)
17:12:27:664 binding for cluster 0x0008 of 0x14B457FFFE7B1011 exists (verified by reporting)
17:12:38:727 max transmit errors for node 0x14B457FFFE7B1011, last seen by neighbors 13 s
17:12:38:987 0x14B457FFFE7B1011 error APSDE-DATA.confirm: 0xA7 on task
17:12:38:988 max transmit errors for node 0x14B457FFFE7B1011, last seen by neighbors 13 s
17:16:46:039 max transmit errors for node 0x14B457FFFE7B1011, last seen by neighbors 86 s

PeterRemnemark, 2020-01-01:
Not sure if I can draw any conclusions, since I do not have a deep knowledge of the protocol nor the real impact of the WiFi network interference. However, I get more lights that gets "offline" or irresponsible and when making a WiFi analyse I see that one neighbour have selected ch 13 with a max signal strength towards my property (and devices) of 33%. My zigbee ch is 25. Very close in Mhz if I undersand correctly. 
I did also reduce the power of my own AP to low (on Unifi), but for some odd reasons my problem escalated to include more lights to not working properly.

Anyone that has seen this before and found a good way of solving it?

Thanks,
Peter

MikaelHoogen, 2020-01-02:
As you mentioned very similar to my problem. In my case the problem is affecting nodes that not have the gateway in it's neighbour table. So it's affecting nodes where the commands need to be routed through another node.

In short. Nodes that have a direct link when you look in deconz are working. Nodes that don't have have a direct link is becoming unresponsive.

MikaelHoogen, 2020-01-02:
Do you have a multi vendor network?

ginkel, 2020-01-02:
> In short. Nodes that have a direct link when you look in deconz are working. Nodes that don't have have a direct link is becoming unresponsive.

I am almost certain that this is a deCONZ problem. Last night a SPZB0001 became unresponsive (although it still sent status update). From the logs I could see that deCONZ did not even try to send data to  the node. One deCONZ restart later everything was working again.

PeterRemnemark, 2020-01-02:
> Do you have a multi vendor network?

Yes, I've both bulbs from IKEA and Innr and other devices from Xiaomi (but those do not route, though). I've not made similar analyse as you, but it does sound as an potential "issue" that if one is affected the rest behave badly in that chain. Will make further checks. 
I had forgot that I had the Xiaomi hub plugged in, only used as a night light. I disconnected it to see if it went better. I cannot say it removed my issue, maybe made it better, but I restarted deconze and alos power cycled some of the lights as well so it is not possible to conclude the issue to the hub that had zigbee network impacts.

What I can tell is that this setup has been working for a long time and it did start after the fw and software update that I did in the same time as I extended the amount of bulbs.

PeterRemnemark, 2020-01-04:
I thought I got a solution, to update to .73 version of deconz. But that did not help. The update, power cycle, removal of Xiaomi Hub, and restart of deconz made my zigbee network better, but the main issue was not solved. 

Yesterday the **0x000B3CFFFEFFFDDE** device was the one that did not respond and thankful (perhaps) I recreated deconz container (when updated) with the debug flags set to 1 so I got some more info as of below. To me it does not say anything clearly, but perhaps for someone that knows the protocol it might be useful? The issue was "self healed" i.e. the device/bulb went into unknown state and then to fully responsive and working. 

```
,22:55:27:251 ZDP Mgmt_Lqi_rsp zdpSeq: 188 from 0x14B457FFFE6BD25D total: 15, startIndex: 3, listCount: 3
,22:55:27:251     * neighbor: 0x000D6FFFFE9C1858 (0x1DA0), LQI: 47, relation: 0x02 rxOnWHenIdle: 1
,22:55:27:252     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 32, relation: 0x02 rxOnWHenIdle: 1
,22:55:27:252     * neighbor: 0x14B457FFFE799803 (0x527E), LQI: 40, relation: 0x02 rxOnWHenIdle: 1
,22:55:27:590 Mgmt_Lqi_req zdpSeq: 189 to 0x14B457FFFE6BD25D start index 6
,22:55:27:590 APS-DATA.request id: 43, addrmode: 0x03, addr: 0x14b457fffe6bd25d, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
,22:55:27:644 APS-DATA.confirm id: 43, status: 0x00 SUCCESS
,22:55:27:645 APS-DATA.confirm request id: 43 -> confirmed, timeout 1578088527
,22:55:27:738 APS-DATA.indication srcAddr: 0x2479, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 247, rssi: -60
,22:55:27:739 APS-DATA.indication request id: 43 -> finished
,22:55:27:739 APS-DATA.request id: 43 erase from queue
,22:55:27:739 ZDP status = 0x00 -> SUCCESS

,22:55:45:561 APS-DATA.request id: 173, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
,22:55:45:591 APS-DATA.indication srcAddr: 0x9d11, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 247, rssi: -60
,22:55:45:591 APS-DATA.indication request id: 172 -> finished
,22:55:45:591 APS-DATA.request id: 172 erase from queue
,22:55:45:592 ZDP status = 0x00 -> SUCCESS
,22:55:45:592 ZDP Mgmt_Lqi_rsp zdpSeq: 240 from 0x14B457FFFE7B22ED total: 14, startIndex: 9, listCount: 3
,22:55:45:592     * neighbor: 0x14B457FFFE7B191B (0xAA3A), LQI: 73, relation: 0x02 rxOnWHenIdle: 1
,22:55:45:592     * neighbor: 0x14B457FFFE71EF2E (0xAF01), LQI: 52, relation: 0x02 rxOnWHenIdle: 1
,22:55:45:592     * neighbor: 0x14B457FFFE664C5A (0xC1BE), LQI: 92, relation: 0x02 rxOnWHenIdle: 1
,22:55:45:635 APS-DATA.confirm id: 173, status: 0x00 SUCCESS
,22:55:45:635 APS-DATA.confirm request id: 173 -> erase from queue
,22:55:45:686 aps request id: 173 finished, erase from queue

,22:55:45:863 APS-DATA.request id: 177, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
,22:55:45:909 APS-DATA.confirm id: 177, status: 0x00 SUCCESS
,22:55:45:909 APS-DATA.confirm request id: 177 -> erase from queue

,22:55:53:000 aps request id: 47 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0x000B3CFFFEFFFDDE (0xFFFD)
,22:55:54:079 aps request id: 47 finished, erase from queue

```

At this time I started to force the device to been put into unknown state, i.e. trigged on/off and dimming the bulb

```
22:56:25:445 0x000B3CFFFEFFFDDE error APSDE-DATA.confirm: 0xA7 on task
,22:56:25:445 APS-DATA.confirm id: 122, status: 0xA7 NO_ACK

,22:58:44:762 delay sending request 245 dt 1 ms to 0x000B3CFFFEFFFDDE, cluster 0x0008

,22:58:48:662 APS-DATA.request id: 15, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 6 tx.options 0x04

,22:59:01:416 ZDP Mgmt_Lqi_rsp zdpSeq: 17 from 0x14B457FFFE6DFB0E total: 17, startIndex: 6, listCount: 3
,22:59:01:416     * neighbor: 0x000B3CFFFEF18459 (0x327A), LQI: 52, relation: 0x02 rxOnWHenIdle: 1
,22:59:01:416     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 124, relation: 0x02 rxOnWHenIdle: 1
,22:59:01:416     * neighbor: 0x000D6FFFFEFD4BBD (0x394E), LQI: 59, relation: 0x02 rxOnWHenIdle: 1

23:00:29:345 ZDP Mgmt_Lqi_rsp zdpSeq: 5 from 0x14B457FFFE505366 total: 16, startIndex: 6, listCount: 3
,23:00:29:345     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 44, relation: 0x02 rxOnWHenIdle: 1
,23:00:29:346     * neighbor: 0x000D6FFFFEFD4BBD (0x394E), LQI: 63, relation: 0x02 rxOnWHenIdle: 1
,23:00:29:346     * neighbor: 0x14B457FFFE6FE2A1 (0x5CE3), LQI: 46, relation: 0x02 rxOnWHenIdle: 1
,
,23:00:31:246 discovery for zombie 0x000b3cfffefffdde dropped, last try was 336 s ago

23:04:38:578     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 129, relation: 0x02 rxOnWHenIdle: 1
,23:04:39:862 discovery for zombie 0x000b3cfffefffdde dropped, last try was 585 s ago
,23:04:55:565     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 57, relation: 0x02 rxOnWHenIdle: 1
,23:04:57:360 send NWK_addr_req to 0x000B3CFFFEFFFDDE, last seen 0 s, last seen by neighbors 1 s
,23:05:10:463 ZDP active ep request to 0x000b3cfffefffdde
,23:05:12:680     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 102, relation: 0x02 rxOnWHenIdle: 1
,23:05:18:401 APS-DATA.request id: 218, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 6 tx.options 0x04
,23:05:19:163 delay sending request 223 dt 1 ms to 0x000B3CFFFEFFFDDE, cluster 0x0004`
[... 4 lines cut ...]
```
[... cut ...]
,23:06:03:415 APS-DATA.request id: 90, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 0 len: 6 tx.options 0x04
,23:06:03:507 ZCL attribute report 0x000B3CFFFEFFFDDE for cluster 0x0008, ep 0x01
,23:06:03:554 	0x000B3CFFFEFFFDDE force poll (2)
,23:06:05:387     * neighbor: 0x000B3CFFFEFFFDDE (0x3735), LQI: 214, relation: 0x00 rxOnWHenIdle: 1
,23:06:06:366 APS-DATA.request id: 123, addrmode: 0x03, addr: 0x000b3cfffefffdde, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00

```

corneyl, 2020-01-06:
I seem to have the same problem in my network! Randomly some lights don't turn on/off.

MikaelHoogen, 2020-01-07:
What type of IKEA bulbs do you have? My GU10 bulbs seems to be equal to trouble.

jtebbens, 2020-01-07:
Using deconz 2.05.69 at the moment and ok with ikea E27, E14 and the power sockets. I was using the 2.05.71 and all the ikea stuff kept dropping out.

PeterRemnemark, 2020-01-08:
> What type of IKEA bulbs do you have? My GU10 bulbs seems to be equal to trouble.

I’m using some E27 opal 1000lm, few E14 opal 400lm and a bunch of GU10 ww 400lm. Couple of Innr RB148T. 

I thought the .71 worked better than .70. Not sure if you can go further below than .70 since I made a fw upgrade with this release.

ginkel, 2020-01-08:
What I can also observe is that devices that connect through a TRADFRI light as router become very unreliable to reach, especially the SPZB0001. The issues vanish as soon as the affected devices connect directly to the ConBee II (which I can hardly influence or force, though).

itsjakey, 2020-01-18:
Same experience here with Tradfri bulbs, mainly issue with GU10, but also less so with an e27 bulb.

I’ve only recently started using Deconz with a raspberrypi and raspbee, but so far I must say it’s frustrating. 
I have 3 bulbs in my hallway group, and you are lucky to get all 3 to come on anymore. 
Really disappointed at moment.

Has anyone had ikea bulbs work reliably?

PeterRemnemark, 2020-01-21:
Until half a year ago, it was rock solid. Everything worked as expected. But then something happen and I cannot say what since I changed a number of things, including adding a decent number of bulbs.

PeterRemnemark, 2020-01-29:
Have for a number of days, been running 2.05.67 version. 
There has been som minor glitches, but not the same as previous. This morning and last night everything has worked as expected. Cross my fingers, hope it remains. 
Side note. Have not downgraded the conbee firmware, not sure if needed, but that would be the next logical step if the stability does not remains.

closed 2020-06-05 by stale[bot]