dresden-elektronik/deconz-rest-plugin #2258: Unable to use REST Apis - ConBeeeII on Hassio

flykenobi, 2019-12-29:
Hello everyone,
I'm trying to use REST apis with my conbee II running on Hassio. My Deconz integration in Home Assistant works flawlessly, I have no problem at all, but, when I try to use REST APIs I don't get responses from the stick. Following the procedure described in the documentation I have this reply doing GET request to https://dresden-light.appspot.com/discover:
`[
    {
        "id": "00212EFFFF0486F9",
        "internalipaddress": "172.30.33.2",
        "macaddress": "00212EFFFF0486F9",
        "internalport": 40850,
        "name": "Phoscon-GW",
        "publicipaddress": "xxx.xxx.xxx.xxx" 
    }
]`
Now..trying to get the API key, I have no replies using the 172.30.33.2 address nor using the Hassio server address with the indications taken from the deConz's Hassio plugin page:

Host: core-deconz
(API) Port: 40850
WebSocket Port: 8081

My goal is to use the node-red-contrib-deconz palette in NodeRed...any suggestion would be highly appreciated!

robertlandes, 2020-01-02:
Latest deCONZ add-on in Hass.io uses ingress and external port bindings are disabled. Make sure, that you configured external ports to be used:
[image]

I you are using the official Node-Red add-on inside Hass.io running on the same device, you should be able to connect through the internal network from one container to the other.

I.e. using host name `core-deconz` from Node-Red.

flykenobi, 2020-01-04:
Thank you Robert, the settings were right, I had to push "reset to default" and set again to make it work...

closed 2020-01-04 by flykenobi