dresden-elektronik/deconz-rest-plugin #2259: [Device Support Request] Smarthings Motion Sensor (GP-U999SJVLBAA)

shnhrrsn, 2019-12-29:
This was originally requested in #1260 and there appears to have been work done in 
d2c33bd, however it’s still not working and there seems to be a lot of confusion in original issue (which was closed), so reopening with fresh screenshots.

[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2019-12-30:
I'm afraid you haven't read the clusters before taking the screenshots. What specifically is not working?

shnhrrsn, 2019-12-31:
Ahh sorry, I'll try it again tomorrow — need to unpair with my other hub first.

As far as not working, it’s not appearing in the rest API at all under `/sensors` (I’ve checked `/lights` and `/switches` to be safe, but nothing there either).

SwoopX, 2020-04-29:
@shnhrrsn Sry, must have overlooked this one completely. Still an issue? Otherwise, I'd take another look.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix