dresden-elektronik/deconz-rest-plugin #2260: Ubisys J1 Shutter, how can i calibrate the J1 via phoscon?

overdrive69, 2019-12-30:
Is it possible to calibrate the Ubisys J1 via phoscon?
If Yes, how?
If No, how should I calibrate it?

regards

SwoopX, 2019-12-30:
Taken from **window_covering.cpp**:

* Ubisys Shutter Control J1
 *
 * http://www.ubisys.de/downloads/ubisys-j1-technical-reference.pdf
 * page 18, chapter 7.2.5.1. Calibration
 *
 * Step 1
 *    In order to calibrate the device, first choose the appropriate device type.
 *    0 = Roller Shade Lift only, ..., 6 = Shutter Tilt only, ..., 8 Tilt Blind Lift & Tilt
 *    Write attribute 0x10F2:0x0000 (“WindowCoveringType”) accordingly.
 * Step 2
 *    Prepare calibration by setting these values:
 *    Write attribute 0x10F2:0x0010 (“InstalledOpenLimitLift”) as 0x0000 = 0cm.
 *    Write attribute 0x10F2:0x0011 (“InstalledClosedLimitLift”) as 0x00F0 = 240cm.
 *    Write attribute 0x10F2:0x0012 (“InstalledOpenLimitTilt”) as 0x0000 = 0°.
 *    Write attribute 0x10F2:0x0013 (“InstalledClosedLimitTilt”) as 0x0384 = 90.0°.
 *    Write attribute 0x10F2:0x1001 (“LiftToTiltTransitionSteps”) as 0xFFFF = invalid.
 *    Write attribute 0x10F2:0x1002 (“TotalSteps”) as 0xFFFF = invalid.
 *    Write attribute 0x10F2:0x1003 (“LiftToTiltTransitionSteps2”) as 0xFFFF = invalid.
 *    Write attribute 0x10F2:0x1004 (“TotalSteps2”) as 0xFFFF = invalid
 * Step 3
 *    Enter calibration mode:
 *    Write attribute 0x0017 (“Mode”) as 0x02.
 * Step 4
 *    Send the "move down" command and "stop" after a few centimeters.
 * Step 5
 *    Send the “move up” command. When the device reaches its top position,
 *    J1 will recognize the upper bound.
 * Step 6
 *    After J1 has reached the top position and the motor has stopped, send the “move down” command.
 * Step 7
 *    After J1 has reached the lower bound and the motor has stopped, send the “move up” command.
 *    J1 will search for the upper bound. Once the top position is reached,
 *    calibration of the total steps in both directions is complete.
 * Step 8
 *    In case of a tilt blind set attribute 0x10F2:0x1001 and 0x10F2:0x1003 to the time it takes for a lift-to tilt
 *    transition (down) or a tilt-to-lift transition (up), respectively. Otherwise proceed with the next step.
 * Step 9
 *    To leave calibration mode, clear bit #1 in the Mode attribute, e.g. write attribute 0x0017 as 0x00.
 */

overdrive69, 2019-12-30:
Thanks, I already read this. But is there a way to set this up with phoscon?

overdrive69, 2019-12-30:
ok it seems this is the solution.
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/813

SwoopX, 2019-12-30:
Not in Phoscon, but good catch with the REST call. Let me know if it worked allright so I can keep that one tight for the next time I have to calibrate mine ;)

closed 2020-05-06 by stale[bot]

jan666, 2020-08-15:
Hello

I just paired my J1 (using an Awning) with deCONZ. Controling via deCONZ works, but not via homekit. I wanted to calibrate it like described here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/813 but I dont have a sensor - just a light. Did something change here? How to calibrate?