dresden-elektronik/deconz-rest-plugin #2261: Aqara Switch ZNCZ12LM registered as light

D34DC3N73R, 2019-12-30:
Using ConBee II with deCONZ docker image on Ubuntu 18.04. I went through the steps to add a new switch, but it's automatically added as a light lumi.plug.maus01. Let me know if you need any other info.

ConBee II
Version: 2.05.71
Firmware: 264A0700

BrianPuccio, 2020-01-05:
Same with a ConBee II with deconz installed via apt on a clean Debian 10 (buster) install.

Smanar, 2020-01-06:
But ZNCZ12LM is a plug, not a switch ? Have you a link with the product ?

BrianPuccio, 2020-01-06:
@Smanar: Correct. I believe OP misspoke. I added it myself as a light.

https://www.amazon.com/dp/B07CJ2MM6Z/

teich, 2020-01-07:
Screenshot of Deconz with the plug:
[image]

And how how assistant sees it in the power sensor (switch shows up fine):
[image]

Note that Home Assistant also doesn't see any current or voltage in the sensor. LMK if I should file a separate issue for that!

D34DC3N73R, 2020-01-08:
Yes, I was under the assumption that "plugs" would fall under the switch category rather than the light category. I have mine hooked up to a fan so it's a little odd that it's registered as a light.

SwoopX, 2020-01-13:
Pretty sure that this is due to the fact that it identifies itself as smart plug...

D34DC3N73R, 2020-01-13:
Why are smart plugs categorized as lights as opposed to switches?

ebaauw, 2020-01-13:
From a ZigBee perspective, smart plugs and wired switches, used to "smartify" dumb devices, look, smell, feel, and behave like smart lights.  They expose server (blue) _OnOff_ and _Level Control_ clusters, to _receive_ commands.  Therefore they're exposed as `/lights` resources.

Wireless switches, used to control smart lights, on the other hand, look, smell, feel, and behave like sensors.  They expose client (grey) _On/Off_ and _Level Control_ clusters to _send_ commands (except for the Xiaomi wireless switches before the Opple, who use server clusters to _send_ attribute reports).  These are exposed as `/sensors` resources.

ronaldheft, 2020-01-14:
Just got one of these switches and I'm seeing the same thing. @D34DC3N73R are you seeing anything register in HomeAssistant outside of the power sensor? I only see the power sensor.

ronaldheft, 2020-01-14:
> Just got one of these switches and I'm seeing the same thing. @D34DC3N73R are you seeing anything register in HomeAssistant outside of the power sensor? I only see the power sensor.

Correction, a restart of Home Assistant and the plug shows up. I am, however, seeing the same results and the power sensor appears to always be showing `-1`.

SwoopX, 2020-02-18:
Newly released version .73 might have a fix for the power measurement. Can you try it out?

mcallan83, 2020-02-25:
@SwoopX, just tried on .73 and it doesn't seem to be fixed.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

Doommius, 2021-08-11:
There still seems to be some issues with this (ZNCZ12LM) but might also concern SP-EUC01, There is a memory feature where it remembers the state it was in prior to a blackout or similar power outage event.

I've requested more information from the manufacture about the endpoint needed for this functionality.