dresden-elektronik/deconz-rest-plugin #2262: Selecting contact Sensor.

Luigi-01, 2019-12-30:
Everything runs fine for the last few month.
Today I added one motion sensor and a contact sensor. 
When edeting a group the motion sensor was selectable. 
However I can not select the contact sensor.
I would like to use the contact sensor to control some lights. 
The contact sensor is an xiaomi contact sensor.
Greetings,

Smanar, 2019-12-30:
I think this feature is not possible yet in Phoscon.
You need to use rule or an application that work with the API.

Luigi-01, 2019-12-31:
Thank, I have solved the issue by using Domoticz.
It is very strange that very fancy items are incorporated and the Basic things are forgotten.
I do not have a fancy setup there the difficult thing are done by Domoticz.
But Basic things as a Dimmer and an Open/Close sensor are NOT present.
Greetings,
P.S. What is going on with the Phoscon App, Error 404 site not found.
       Has Dresden Elektronik some issues?

Smanar, 2019-12-31:
Lol, yep, I think too Phoscon need some improvement, but this app is closed source, and changes are not frequent.

closed 2020-05-06 by stale[bot]