dresden-elektronik/deconz-rest-plugin #2265: Deconz lose connection to Trådfri bridge after v. 2.05.69

Michaelnorge, 2019-12-30:
My hardware: Raspberry Pi3, ConbeeII USB and ioBroker

I updated Deconz to version xx.70, xx.71 and xx.72.
I'am using "curtains KADRILJ" ( 904.081.32) since a couple of weeks and everything worked well.
But after I updated to a version higher than 2.05.69 deconz losing the connection to the IKEA bridge ( 403.378.06).
The log in ioBroker showed this:

2019-12-28 22:28:31.498  - warn: tradfri.0 Lost connection to gateway
2019-12-28 22:28:46.503  - info: tradfri.0 Connection to gateway reestablished

The command was not sent to the curtains.

So I had to go back to 2.05.69.

Greetings
Michael

olemr, 2019-12-30:
Excuse me asking, but why use the Ikea Gateway when you have deCONZ?

Michaelnorge, 2019-12-30:
Because sometimes I¨m losing conections to some devices.
I guess I reached the maximum of devices.  randomly there are units I can not reach anymore.
The max devices is indicate by 200 - but I think that there is more than 1 dataset per device. (xiaomi thermometer has for example at least 4 datasets) what means 4 devices i guess,,,
However: I am using both conbee as well as IKEA Gateway.

olemr, 2019-12-31:
I see. I also keep my Ikea Gateway with one bulb and one remote just to keep track of 'what is happening'. If I ever hit the 200 node limit, I could setup a second deCONZ network, but that hasn't happened yet. I did have some issues with 16 E27 & 12 GU10 Ikea bulb becoming unreachable, but I moved them over to a HUE bridge. I feel things improved, but I still get the occasional dropout and have to power-cycle the bulb(s).

closed 2020-05-06 by stale[bot]