dresden-elektronik/deconz-rest-plugin #2266: Strange - why is the remote control not visible?

Michaelnorge, 2019-12-30:
I connected a remote control (up/down) from insite a package of the curtain "KADRILJ" to conbee2.

The connection was not established after three minutes, so I aborted it.
But than I saw that remote control in the list objekt-list of ioBroker, and it is working!!

Strange, because the remote control is not shown up in the phoscon app...

Anybody who can explain what happened?

Greetings
Michael 
[image]

Smanar, 2019-12-30:
Some device are not supported by phoson, but are full working in the API.

olemr, 2019-12-30:
You will find and link to it here: 
[image]

Michaelnorge, 2019-12-30:
Thank you all for your informations!

closed 2019-12-30 by Michaelnorge