dresden-elektronik/deconz-rest-plugin #2269: Deconz showing wrong state on Aqara door/windows sensors

HaaVeeEnn, 2019-12-31:
Version 2.05.72 / 12/12/2019
Conbee 2 firmware 264A0700
Hass.io version 0.103.5
Some first times it worked correctly but after it stopped.
Even if i open and close window state not changes, refreshed staying "21 minutes ago"
[image]
[image]
[image]
[image]
[image]
[image]

ebaauw, 2019-12-31:
Did you replace the sensor's battery?

HaaVeeEnn, 2019-12-31:
new device with 100%

D34DC3N73R, 2020-01-01:
I have the same sensor with conbee II but running deconz in docker on ubuntu 18.04 and it seems to be working fine, Strange that yours is reporting vibration and doesn't have the version.
[image]

robertlandes, 2020-01-02:
Did you try to press the little button on the sensor (blue LED should blink once or twice) ? That should actually trigger an update to the state more or less immediately.

HaaVeeEnn, 2020-01-02:
i repaired all those sensors who not shown version, and  now it works fine. but definitely there is some kind a problem why it wasn't paired correctly with first time

TilBumble, 2020-01-27:
After updating from 2.05.71 to 2.05.72, my sensors still seem to work, but they show an additional line in their details page, as in the opening post, stating "Vibration {vibration}".

This refers to all my sensors:
- TRÅDFRI Motion sensor (2 pcs.)
- Xiaomi Multi Sensor 20161129 (2 pcs.)
- Xiaomi Window / Door Sensor 20161128 (2 pcs.)
- Xiaomi Water Sensor 20170721 (1 pcs.)

In Openhab 2.5.1-2, no new channels were created.

The only other change i noticed is that my Tradfri 5-button-remotes have disappeared from the start page, but are working and shown under "Switches".
(I had no lights directly assigned to them)

As the system seems to be functional, both issues are only minor flaws for me, but perhaps might be interesting for some of you.

Raspberry Pi 3 B+
ConBee 26330500
headless on openhabian

closed 2020-06-02 by stale[bot]