dresden-elektronik/deconz-rest-plugin #2270: How do you change 'power failure' modes and fade durations?

Webunity, 2020-01-01:
I have the Docker Deconz running, all seems to work fine; paired about 10 Hue bulbs but i would like to change their power failure settings and for example how long it takes for a bulb to fade from one setting to another (currently it is a second i think).

I cant seem to find these settings in the GUI, but maybe i have to change it in the config in the docker image itself? The URL i am looking at starts with /pwa/index.html?...

robertlandes, 2020-01-02:
Inital power on state is IMO currently not supported in deCONZ (only through HUE bridge). Fade duration (transition time) can be set on the edit tab in scenes configuration.

[image]
[image]

ebaauw, 2020-01-02:
See #1014.  With some tweaking of `general.xml` it's possible to set the poweron state of Hue lights through the deCONZ GUI.

closed 2020-05-09 by stale[bot]