dresden-elektronik/deconz-rest-plugin #2272: Feature Request: "Reset node" function in REST API

dandrzejewski, 2020-01-01:
I sometimes find my self needing to do a "Reset Selected Node" in deCONZ. It would be nice if there were an API for this.

That way, I could enumerate the devices, see which ones are unresponsive, and then issue a reset command to those particular nodes.

SwoopX, 2020-01-02:
What exactly do you want to do with that? You could delete the device in Phoscon and then re-join it but that does nit reset the device itself. Not sure if that is even technically possible from a device perspective.

dandrzejewski, 2020-01-02:
Maybe I have the wrong idea of what that function does in deCONZ. Can you explain it to me?

SwoopX, 2020-01-02:
Sorry, should have read more carefully and totally forgot about that option. Good question what it exactly does. Haven't found anything in the manuals...

From a network perspective, it looks like everything's freshly read in without a re-join. Could be wrong on it though.

dandrzejewski, 2020-01-03:
To be honest, the manual leaves a bit to be desired.  You can press "0" on a node apparently and it does... something.

closed 2020-05-09 by stale[bot]