dresden-elektronik/deconz-rest-plugin #2274: (ConBee II) Phoscon app unreachable after dietpi update.

msantagata, 2020-01-02:
Hi Folks, hope you can help me!
I have a ConBee II dongle connected to a RPi 3A+ running dietpi (a stripped down version of Raspibian). Since it has no desktop, I have installed deconz "headless".
Everything was working fine until I updated dietpi to the latest version (6.27.2): the dongle seems to be unreachable, and the phoscon app cannot be loaded.
A few notes:

- If I try to run manually `/usr/bin/deCONZ -platform minimal --http-port=80`(that was working fine before the update) I get a few warnings/errors

```
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints() 
PROTO: CRC error
PROTO: CRC error
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
```

- I have tried to reconnect the dongle, and change the USB port, but with no luck

- If I try to update the firmware with
`sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF`
I get :
`/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version 'GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)`

I understand that probably something's changed on dietpi side, but do you have any advice on how to fix this?

Thanks!

itsjakey, 2020-01-13:
I have a raspbee and had a similar issue after updating dietpi.
I pointed it direct at /dev/serial0 which seemed to resolve it.

closed 2020-05-20 by stale[bot]

msantagata, 2020-05-24:
> I have a raspbee and had a similar issue after updating dietpi.
> I pointed it direct at /dev/serial0 which seemed to resolve it.

Hello @itsjakey , can you tell me how to do that? Thanks!