dresden-elektronik/deconz-rest-plugin #2275: How to integrate/connect a unsupported device?

Monnem, 2020-01-02:
Hi,
I got some motion sensors from the big A called:
"pegtopone otion Sensor ZigBee" (yes, without m in motion) ;-)
The sensor itself works as the LED lights when motion is detected.
But I fail to pair it with deconz. Any hints how to do that?

Thanks and regards

SwoopX, 2020-01-02:
Please issue a device support request as described in the wiki.

Smanar, 2020-01-03:
But the first step for this is make the device pairing with deconz.
If you can't include the device, it will be harder, need to check if it use correct zigbee request, ect ...

Hassfabi, 2020-01-11:
Hi Together,
guess I purchased the same motion sensor on Amazon.
Here's what I got via VNC (searched for new sensors via phoscon web interface,  resetted the motion sensor and it joined the network):
[image]
Node info:
[image]
Cluster IAS Zone:
[image]
[image]

If there's anything else you need, please let me know. How long does it usually take until a new device is added?
I'm also wondering if there's a way to use motion sensor already with the help of the binding drop box? Is there anything to read about how to set up a motion sensor, switching on and off a light with the binding drop box?

jensflorian, 2020-01-22:
I am also looking for examples using the binding drop box. So far I have found only this:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/881#issuecomment-431654926

Smanar, 2020-01-22:
The binding alone will be not usefull.

You can try to add yourself the device in the API, if the device is standard take a look on a previous pull request, some device just need 4/5 lines to be added, but for test it s better if you have the device.

closed 2020-05-30 by stale[bot]