dresden-elektronik/deconz-rest-plugin #2276: Allow remapping of baseurl/port

codepitbull, 2020-01-02:
This isn't a bug, more a "recommended addition" :)

Since the deconz-rest-plugin doesn't support ssl natively (and I am not suggesting it should)  I decided to put it behind an nginx reverse proxy.
This works pretty straight forward for simple setups but more complex setups can be hard to achieve.
This is due to the fact that the rest-plugin doesn't support modification of the returned URLs.

In nginx I would normally map all requests to deCONZ using a common path.
All requests to **<nginx-host>/hue/pwa** would go to **<deconz-host>/pwa**.
This works easily for requests being performed directly against the proxy, but returned documents and requests contain URLs generated by deCONZ. 

An example:
When not logged in a request to **<nginx-host>/hue/pwa** would result in a redirect to **/pwa/login.htm**, breaking the rewrite in nginx and resulting in a 404.
Returned documents could be rewritten but this is error prone and fragile since each new release might break these conversions.
Having the ability to change the URLs generated by deCONZ by the simple addition of a prefix would help a lot to integrate well with reverse proxies.

SwoopX, 2020-01-03:
That would be really awesome! Also having trouble with nginx here and would love to see it working smoothly behind it.

msnelling, 2020-01-30:
Agreed. Using it behind Traefik is also not possible as it can't do URL rewriting.

ebaauw, 2020-01-30:
I think you're confusing the REST API (`/api/`...) with the Phoscon web app (`/pwa/`...).

> This is due to the fact that the rest-plugin doesn't support modification of the returned URLs.

The only URL that the REST API returns is the link to `description.xml` in the UPnP announcement.  Technically, you can put the REST API behind a reverse proxy without any issue, telling the REST API client to connect to the proxy hostname:port instead of to the gateway.  If you want to change the base URL, you obviously would need to define a way to to return the base URL in the UPnP and appspot announcements.  However, in addition to that, you would have to change _every_ REST API client to use that base URL instead of `/api`.  We know several Hue API clients that don't even work on a port other than 80, so using a different base URL is out of the question.

The Phoscon web app is just another REST API client running in your web browser.  The version that gets distributed with deCONZ happens to be served from the same web server as the REST API, but the web app can be served from any other web server.  It's "just" a collection of static web pages and JavaScript code.  It's far less of a priority to serve that over HTTPS, as there is no sensitive data there.  The connection to the gateway is made from your browser running the JavaScript.  And like any other REST API client, it's hard coded to the `/api` base URL of the gateway.

codepitbull, 2020-03-02:
Fair enough and I guess this is the wrong forum for my actual intent (you might have noticed I specifically mentioned the pwa-endpoints).
I have no intends to use the REST-API.
I have also no intend to deploy the UI to separate server.
I have a raspberry in a separate VLAN running both, I just want to have SSL-termination in front of it.
Right now I can't see how to get this working in nginx without doing heavy manipulation inside nginx, which has already proven to rather error prone.

Where would I file such a request?

SwoopX, 2020-05-19:
Hm, as I had to renew all my internal certificates, I recalled that I still wanted to get all the communication SSL-terminated as well. Played around with that just a bit and on the first brief look, it seems to be working.

Served Phoscon natively via nginx and proxied any connections to `/api` to localhost. @codepitbull isn't that what you were after?

codepitbull, 2020-05-20:
@SwoopX that sounds like what I want.
Can you share your configs?

SwoopX, 2020-05-20:
Sure. That should be the relevant parts. I've deconz running on port 8080

`lrwxrwxrwx 1 root     root       25 May 19 21:52 hue -> /usr/share/deCONZ/webapp/`

```
    location /hue {

        allow       192.168.1.2;
        deny        all;

        #root /var/www/html/webapp;
        root /var/www/html;
        index index.html;

    }

    location /api/ {

        proxy_set_header        Host $host;
        proxy_set_header        X-Real-IP $remote_addr;
        proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header        X-Forwarded-Proto $scheme;
        proxy_http_version      1.1;

        proxy_pass              http://127.0.0.1:8080/api/;

    }
```

However, `https://192.168.1.10/hue` runs into a not found due to a referrer issue. Regardless, from `https://192.168.1.10/hue/pwa/login.html`, everything seems to run fine (and really fast). Note that I just spend a couple of minutes for this, so it might be a bit far from being perfect.

closed 2020-06-17 by stale[bot]