dresden-elektronik/deconz-rest-plugin #2277: Alexa does not detected Conbee gateway

funbben, 2020-01-03:
Hello, 
I run a conbee II on a TV box with armbian / Debian Stretch (Version : 2.05.71 / 17/11/2019 - Firmware : 26490700)

There is also a Jeedom server installed on the same box. I've set jeedom to run on port 81. and launch a new instance of deconz on port 80 (Jeedom deconz is running on port 8484, And I don't know how to get control over this instance, but controlling device from port web app on 80 is fine)

I used "Upnp Analyzer" that show that deconz expose on port 80 (see below) but no success to get Alexa to find the gateway. 
Also, It seems weird that the serial are "0000000000000000" 

Any idea, on what to look for to solve this ?

```
NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.1.16:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0
GWID.phoscon.de: 0000000000000000
hue-bridgeid: 0000000000000000
NTS: ssdp:alive
NT: urn:schemas-upnp-org:device:basic:1
USN: uuid:df430cc6-7dcc-4a21-ab9e-cbe6c95ec3b6::urn:schemas-upnp-org:device:basic:1

```

jpawlowski, 2020-01-04:
You will need to install something like diyHue to connect to Alexa.
Note that there will be a conflict for port 80 so it is recommended to install diyHue on a separate device (or virtual instance).

Smanar, 2020-01-04:
deconz is not already an Hue emulator ?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Hue-API-Compatibility

ebaauw, 2020-01-04:
The APIs are very similar but not identical.  Especially the `/scenes` api is very different.  I think Alexa will only work with deCONZ, when deCONZ is running on port 80 (like the Hue bridge).  Even then, I understand support is limited.  Search (closed) issues for more info.

funbben, 2020-01-04:
I did read a lot of closed "Discover Alexa" topic but none seems to five a proper recipe to investigate or make this work. 
The UPnP analyser showed that both UDP on 1900 and TCP on 80 are fine. 

I dont inox what to investigate / solve next. Help needed

Smanar, 2020-01-05:
Nothing to see with your issue but before your tests, you haven't forget to enable app authentication for 60 seconds in phoscon ?

funbben, 2020-01-05:
Yes I did enabled app authentication when trying to pair with Alexa.
Here is the XML :

```
<root xmlns="urn:schemas-upnp-org:device-1-0">
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<URLBase>http://192.168.1.16:80/</URLBase>
<device>
<deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
<friendlyName>Phoscon-GW (192.168.1.16)</friendlyName>
<manufacturer>Royal Philips Electronics</manufacturer>
<manufacturerURL>http://www.dresden-elektronik.de</manufacturerURL>
<modelDescription>Philips hue compatible Personal Wireless Lighting</modelDescription>
<modelName>Philips hue bridge 2015</modelName>
<modelNumber>BSB002</modelNumber>
<modelURL>http://www.dresden-elektronik.de</modelURL>
<serialNumber>000000000000</serialNumber>
<UDN>uuid:df430cc6-7dcc-4a21-ab9e-cbe6c95ec3b6</UDN>
<presentationURL>index.html</presentationURL>
<iconList>
<icon>
<mimetype>image/png</mimetype>
<height>48</height>
<width>48</width>
<depth>24</depth>
<url>hue_logo_0.png</url>
</icon>
</iconList>
</device>
</root>
```

funbben, 2020-01-07:
Anyone ?

Schnuecks, 2020-03-30:
Same here, not workin with alexa currently, tried it many times.
xml is valid with royal philips nothing is found.

TheHaf, 2020-04-03:
I also struggle with this. Phoscon is running on port 80, the description.xml already had Royal Philips Electronics as the manufacturer set, but Alexa won't recognize it. I also tried to connect the Phoscon GW to Hue essentials, that worked right off the bat. Does Alexa maybe only support one Hue Bridge at the same time?
I also double-checked the description.xml on my Hue Bridge and noticed that it was set to "Signify" as the manufacturer. So I also tried to set it like that on my Phoscon GW, but that also did not work unfortunately.

savelkouls, 2020-04-04:
Hey Guys,

I have the same problem...
it's running on an raspberry with home assistant and conbee 2 stick. 
The addon works fine but I can't connect to Alexa.
In my opinion it is really strange that the addon description says: "The addon has no UPnP support". 
Does this mean Alexa doesn't work worh this addon?

Normensa, 2020-04-05:
> 
> 
> Hey Guys,
> 
> I have the same problem...
> it's running on an raspberry with home assistant and conbee 2 stick.
> The addon works fine but I can't connect to Alexa.
> In my opinion it is really strange that the addon description says: "The addon has no UPnP support".
> Does this mean Alexa doesn't work with this addon?

I installed the Conbee2 stick on my Win10 notebook and it is still not possible to connect to alexa.

Do you have any idea what I could try?