dresden-elektronik/deconz-rest-plugin #2278: Deconz - Openhab binding: Presence sensors are not updated in openhab

xrmthomas, 2020-01-03:
I am currently experiencing an issue where suddenly (since this morning) all my presence sensors are refusing to update in OpenHab.

The Phoscon gateway shows them working without issues, but the values are just not updated in openhab.  When I re-add them in Openhab they show an online status, and I get the last-updated value once, but after that nothing happens anymore.

I reconfigured the entire setup, removed and readded the things, re-linked the items, nothing seems to work here.

alessandroceleste, 2020-01-19:
I have the exactely same problem with iobroker.

SimpleSimonLA, 2020-02-02:
Any updates on this? I have the same problem with temperature / humidity sensors.

The values are properly shown in the Phoscon web page or when I manually check via REST API for the sensors. But openhab does not get any updates and I don't understand why :(

gabrielgunnarsson, 2020-03-20:
Hi! I also had this issue but with contact and vibration sensors. 

Found in the Phoscon GUI, in the top right corner that it displayed a link local IP address. This was also the IP displayed when on the things properties in OpenHab.

Not sure if this was the issue, but in any case the issue resolved when I went to the old Phoscon GUI and disabled "Internet Service" under Advanced Settings, and then rebooted the device.

xrmthomas, 2020-03-20:
I just changed the setting in my system and rebooted. I’ll report back when results are in. Thanks a bunch!

xrmthomas, 2020-03-21:
Unfortunately after a couple of hours the issue came back.

I noticed a beta-image on dresden-elektronik's website I hadn't installed yet so I flashed the image this morning and loaded my settings backup. Up untill now it seems to run stable and the issue hasn't showed itself. It also seems the connection issue with one of my bulbs has been resolved as well..

eric1905, 2020-05-18:
I now have the same issue. Phoscon is working but openhab is not updating the values. It is not just my motion sensor it is also the default daylight sensor.

Mimiix, 2020-06-12:
As there is no further response and seems to be related to OpenHAB, please head over to their git.

closed 2020-06-12 by Mimiix