dresden-elektronik/deconz-rest-plugin #2280: Failed to start deCONZ (libQt5)

mario-fliegner, 2020-01-03:
Hi,
I'm trying to get the ConBee II (image **deconz-2.05.72-qt5**) working on my RPI3 (_Linux openHAB2 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux_). 
During the service startup I get the following error in the system log:

`Jan  3 17:53:42 openHAB2 systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.`
`Jan  3 17:53:42 openHAB2 systemd[1]: Unit deconz-init.service entered failed state.`
`Jan  3 17:53:42 openHAB2 systemd[1]: Stopping deCONZ: ZigBee gateway -- REST API...`
`Jan  3 17:53:42 openHAB2 systemd[1]: Starting deCONZ: ZigBee gateway -- REST API...`
`Jan  3 17:53:42 openHAB2 systemd[1]: Started deCONZ: ZigBee gateway -- REST API.`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Network.so.5: no version information available (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version 'GLIBCXX_3.4.21' not found (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by /usr/bin/deCONZ)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)`
`Jan  3 17:53:42 openHAB2 deCONZ[3704]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)`
`Jan  3 17:53:42 openHAB2 systemd[1]: deconz.service: main process exited, code=exited, status=1/FAILURE`
`Jan  3 17:53:42 openHAB2 systemd[1]: Unit deconz.service entered failed state.`

The steps I've performed so far:
`sudo gpasswd -a $USER dialout`
`wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -`
`wget https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.72-qt5.deb`
`sudo dpkg -i deconz-2.05.72-qt5.deb`
`sudo apt install -f`
`sudo dpkg -i deconz-2.05.72-qt5.deb`
`sudo systemctl enable deconz`
`sudo reboot`
`sudo systemctl stop deconz.service`
`sudo nano /lib/systemd/system/deconz.service (changed: httpPort to 8090)`
`sudo nano /boot/config.txt (added: enable_uart=1)`
`sudo systemctl disable hciuart`
`sudo apt-get dist-upgrade`
`sudo reboot`

Any thougths?
Thanks!

Itzamna1981, 2020-01-03:
Okay, looks like a Qt version mix? What's the best way to resolve this?

[image]

manup, 2020-01-04:
You need the ARM package here, since Ubuntu is for AMD64 :)

http://deconz.dresden-elektronik.de/raspbian/beta/

Itzamna1981, 2020-01-04:
Thanks, I've now tried it with the http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.05.72-qt5.deb package instead https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.72-qt5.deb. Unfortunately I get the same errors as before. Did a `sudo apt-get purge deCONZ` to remove the existing module completely and then I followed the same steps as mentioned above to install the **raspbian** version.

manup, 2020-01-04:
Is this a Raspbian Jessie?

    $ cat /etc/os-release

The packages in the above download link are build against Stretch with Qt 5.7.
The Jessie packages can be installed via the APT repository, can you please try the following instructions: 

https://phoscon.de/en/conbee2/install#raspbian

Itzamna1981, 2020-01-04:
Yes, it's a Jessie:
`[12:34:21] openhabian@openHAB2:~$ cat /etc/os-release`
`PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"`
`NAME="Raspbian GNU/Linux"`
`VERSION_ID="8"`
`VERSION="8 (jessie)"`
`ID=raspbian`
`ID_LIKE=debian`
`HOME_URL="http://www.raspbian.org/"`
`SUPPORT_URL="http://www.raspbian.org/RaspbianForums"`
`BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"`
`[12:34:25] openhabian@openHAB2:~$`

I tried it with the APT repository and it's coming up now! Looks like working fine :-)
Thanks for your help!

closed 2020-01-04 by Itzamna1981

mat1990dj, 2020-04-07:
Hello, I'm on Jessie and via apt-get I'm only getting .74 version, upgrade via apt-get shows still on latest. Is this normal?