dresden-elektronik/deconz-rest-plugin #2281: Add support for Lupus small ZigBee Temperaturersensor

sergiosaizolmo, 2020-01-03:
It's posible to add this device:
Lupus small ZigBee Temperatursensor (https://www.lupus-electronics.de/shop/en/Alarm-Smarthome/Smarthome-Home-Automation/LUPUS-Small-ZigBee-Temperatursensor-p.html)
Thanks in advance. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-01-03:
Looks like you haven't read the attributes before taking the screenshots. Please do so.

sergiosaizolmo, 2020-01-04:
Sorry for the lapsus. Here is the updated screenshot:
[image]

closed 2020-05-10 by stale[bot]

sbonfert, 2020-07-07:
I got the same device, however, it does not show up in phoscon. I therefore looked at the device properties in deCONZ. Most notably, the manufacturer name seems to differ for my device.
Is it necessary for you to add support for my device or am I configuring it wrong?

[image]
[image]
[image]