dresden-elektronik/deconz-rest-plugin #2282: Add support for Illuminize Wall Switch

AppleCarrotFlame, 2020-01-03:
Please add support for the Illuminize wall switch with 4 functions (on/off, cold/warm, brightness and colour). 

[Web page with device info](https://www.iluminize.com/de/shop/led-steuerung/product/540-511-577-zigbee-wand-dimmer-schalter-rgb-wei%C3%9F-duo-wei%C3%9F-led-1-zone.html)

[image]

[image]

AppleCarrotFlame, 2020-01-06:
I managed to assign a few bindings via the Bind Dropbox. A single lamp (via IEEE address) can now be switched On/Off (cluster 6) or be dimmed (cluster 8). I have yet to find out how to manage groups via deCONZ, because I think it'd be more manageable in the future. 

The device still is not announced via the API to phoscon or to HASS.io.

Shamal2008, 2020-01-10:
I would also really appreciate the support of this device. Is there a chance at the moment to get this switch in the deCONZ Webapp?

closed 2020-05-16 by stale[bot]