dresden-elektronik/deconz-rest-plugin #2283: IKEA Fyrtur - Can't be controlled from Phoscon or Home-Assistant

IDmedia, 2020-01-04:
I've recently brought a Conbee II and migrated all lights, sensors and switches and it works great for the most part. However, I'm really struggling with my IKEA Fyrtur blind (I only have one).

I'm running Home-Assistant on a Raspberry Pi (Hassio) and the official deCONZ addon. Both on latest available version. I have no issues pairing the blind using the "Lights" category. I assume that everything is ok at this point as it even lists the version number (22009631) correctly. After getting it to show up in Phoscon I'm able to control it for a few seconds/minutes. I think the control is really counter-intuitive, but when I move the light-slider in the group the blind is in (the only item), then the blind moves. However, after a few minutes it doesn't respond anymore.

In Home-Assistant I can see the blind, but only the STOP- and DOWN-buttons are available/black. The UP-arrow is grayed out for some reason. The status in general is all wrong as it may report the blind as closed even though it's open and the "X minutes ago" timer just keeps on increasing so it seems it looses connectivity right after pairing.

I have a IKEA Floalt light on the ceiling right next to the blind and I also have two IKEA power sockets to the left and right of it, so I can't believe it's the lack of nodes. Connecting to VNC/the mesh overview every connection is green. I have also tried with and without the repeater without any changes. I've also factory reset the blind multiple times and rebooted the Pi completely without that helping solve the issue.

Has anyone been able to get the IKEA Fyrtur to work properly in Home-Assistant? Is there anything I can do to debug this further? It would be really unfortunate to have to return the Conbee II and go back to the IKEA gateway just because of one device not working.

olemr, 2020-01-04:
My experience as well. I'm just about to test if it works through 'groups' in the REST API. This is the group that is the open/close remote, so I guess it does, but controlling individual or PWA created groups of blinds, only work for a very short period of time after pairing.
Hopefully this will be resolved in the future.

ebaauw, 2020-01-04:
> I'm just about to test if it works through 'groups' in the REST API.

The FYRTUR supports groups.  The open/close remote actually sends group commands.

th3cube, 2020-01-07:
> Has anyone been able to get the IKEA Fyrtur to work properly in Home-Assistant? 

Yes

I have a Conbee II Stick but still running with the Firmware Version 26490700. When i first paired the Blind i had some connectivity Problems too. It was Paired over an Osram Plug acting as a Router. When i paired it directly to the coordinator the connectivity Problems where gone.

D3nnisd, 2020-01-08:
I have paired it since the day it came out and it works just fine. So i am not sure if this is you or something else.

IDmedia, 2020-01-08:
> Yes
>
> I have a Conbee II Stick but still running with the Firmware Version 26490700. When i first paired the Blind i had some connectivity Problems too. It was Paired over an Osram Plug acting as a [...]

I've heard similar stories in other issues here... is there a way to force the blind to talk directly to the coordinator? I've connected to VNC and looked at the mesh, but can't find any way of telling the blind to connect directly to the coordinator. Or do I really have to turn off any other smarthome device in order to make sure it connects directly to it? I wonder if it will still work after turning on all the other nodes afterwards again....

IDmedia, 2020-01-08:
> Yes
>
> I have a Conbee II Stick but still running with the Firmware Version 26490700. When i first paired the Blind i had some connectivity Problems too. It was Paired over an Osram Plug acting as a [...]

You are actually correct (unfortunately?). I took down the IKEA Fyrtur blind and put it right next to my Raspberry Pi and the Conbee II and re-paired it. According to the mesh it's now connected directly to the coordinator instead of to my IKEA Floalt light and it works. Now it grays out the correct arrow and I can move it up and down. I just wonder if it will stay connected to the coordinator or someday switch over to another router and stop working?

th3cube, 2020-01-09:
> I just wonder if it will stay connected to the coordinator or someday switch over to another router and stop working?

My experience is that it stays connected to the  coordinator even if there is a router next to the blind.

closed 2020-01-09 by IDmedia

daggr8, 2021-02-05:
I have the exact same error. I've paired it directly with the ConBee II but still the same. Arrow Up is greyed out. 
I've repaired several times. Works in the Devonz GUI via VNC. 
ConBee FW 26680700
Deconz 2.09.01

Any tips?

kev-null, 2021-02-18:
@daggr8 I 'fixed' this by adding the below to my `customize.yaml`:

`cover.office_blind:
    assumed_state: true`

Not perfect but it should stop greying out the up button.

daggr8, 2021-02-18:
> 
> 
> @daggr8 I 'fixed' this by adding the below to my `customize.yaml`:
> 
> ```
> cover.office_blind:
>     assumed_state: true
> ```
> 
> Not perfect but it should stop greying out the up button.

Thanks! I solved it with the help of [this post ](https://community.home-assistant.io/t/ikea-fyrtur-block-out-roller-blind-support/134692/146)

wryest, 2021-08-13:
In case someone else is still struggling with this:
I also had a greyed out up arrow and wasn't receiving any cover percentage state updates.
I followed the instructions in this [post](https://bleepcoder.com/deconz-rest-plugin/398540031/ikea-smart-blinds-fyrtur-and-kadrilj) by @ebaauw and was able to get everything to work properly by binding and configuring attribute reporting.