dresden-elektronik/deconz-rest-plugin #2285: Add support for 3AZB-41-DE

AppleCarrotFlame, 2020-01-05:
This device is made by a company called 3A, and it is a zigbee 3.0 light switch for a single zone which replaces the in-wall switch. The unit is also known under the Nue brand, and retains the legacy on/off functionality of a normal switch when the zigbee signal is not available.

[image]

[image]

[image]

https://www.amazon.de/gp/product/B07GWR3VBX/ref=ppx_yo_dt_b_asin_title_o00_s02?ie=UTF8&psc=1

Smanar, 2020-01-05:
And it don't work natively ?
If you go in cluster 0006, you can change the bulb state ?

AppleCarrotFlame, 2020-01-05:
It works natively (on/off/toggle works fine). It just does not appear in phoscon nor in HASS.io. All my devices that are named 0x.... do not appear.

Smanar, 2020-01-05:
And you can't see it in the API ?
I don't see anything special on your device, I can explain you how to add it in white list, but as it, from my memory all lights have a device created in api list by defaut.

AppleCarrotFlame, 2020-01-05:
This last post gave me the hint (whitelist in API) I needed. I started the search for "lights" in phoscon again, and then switched the light on and off again. In phoscon there suddenly appeared an On/Off Light 19 with unknown manufacturer, but after a few seconds both the manufacturer and model appeared. 

Thanks! This request can be closed.

closed 2020-01-07 by AppleCarrotFlame