dresden-elektronik/deconz-rest-plugin #2287: Gira Switch no longer displayed in Phoscon Device List

kaeferfreund, 2020-01-05:
Hey there,

i have a raspbeeGW with some hue lights, tradfri stuff, aquaria sensors and a couple of gira wall switches. A week ago the switches lost connection to the raspbeegw so I readded them by manually resetting them. After readding they do appear in the switches section, but not in the general group section where the light groups and previous switches are displayed.

As a result I can not assign any lights to the switch which is kind of unfortunate. Andy ideas?

Smanar, 2020-01-05:
Hello, you can try with the older app (in help/old webapp version), this one display more info than in phoscon, perhaps you will find it here.

Or you can take a look directly in the API, and delete them before pairing.

kaeferfreund, 2020-01-05:
Thanks @Smanar that worked for me. As soon as I assigned a light in the old Webapp, the switch appeared in the new gui again.

closed 2020-05-13 by stale[bot]