dresden-elektronik/deconz-rest-plugin #2289: Bosch ZigBee Wireless Motion Detector RFDL-ZB-MS

geelenbert, 2020-01-06:
I tried to pair a couple of Bosch zigbee motion detectors.

I was hoping it would work out of the box as the devices should support Zigbee HA1.2:
[https://zigbeealliance.org/zigbee_products/radion-tritech-zb-4/](url)

For the Zigbee2mqtt project there has also been a request:
[https://github.com/Koenkk/zigbee2mqtt/issues/769](https://github.com/Koenkk/zigbee2mqtt/issues/769)

It pairs ok, but not shure what i need to do to get it working.
Ultimate i want to connect it to Home Assistant.

[image]

SwoopX, 2020-01-06:
Can you please provide additional info as described in the wiki? Also, please make sure having the attributes read before taking the screenshots.

geelenbert, 2020-01-09:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

geelenbert, 2020-01-20:
Is there any insigt in when the next build will be done?

I don't want to be pushing anyone, but i just want to know when the next build will be. The last one was from over 2 months ago, and i'm ready to test the fix.

SwoopX, 2020-01-24:
> The last one was from over 2 months ago, and i'm ready to test the fix.

You may want to compile the current version yourself ang give it a try?

geelenbert, 2020-01-27:
I would like to do this, but it's not that easy.
My setup is running deconz in as Hass.io addon. 
I can run deconz in a Ubuntu VM for testing.

But what should be te expected behaviour?

SwoopX, 2020-01-27:
Well, from your initial post, I conclude you do not see any sensors exposed through deconz' REST API and therefore also not in Hass.io.
Since the updates have been merged, but .73 is not available yet, compiling the plugin and doing the pairing should expose those sensors now. Pretty much straigh forward.

jensflorian, 2020-01-28:
Does anyone know whether the RFDL-ZB-MS is  to the model sold as Smart Home motion sensor https://www.bosch-smarthome.com/uk/en/shop?url=products/smart-system-solutions/motion-detector ?

geelenbert, 2020-01-28:
So i compiled the latest version and runned the rest plugin into a virtual machine:

[image]

I removed the old Deconz integration from home assistant.
I started Home assstant again and paired the new Deconz instance. 

All my sensors are now added, but I can't find a new motion sensor.

Am i looking into the wrong place? Or is there anything i'm missing ?

SwoopX, 2020-01-28:
Did the sensor search give you a green indication of a successful pairing process with the Bosch sensor? Have you queried the deconz REST API for all sensors and double checked?

geelenbert, 2020-01-28:
I paired the sensor again and now it shows up in Home Assistant.

I enabled the light indicator on the device to see if motion is detected.

But the state of the sensor does not change in Home Assistant:

[image]

Any suggestions ?

SwoopX, 2020-01-28:
Hm, can you double check in deconz GUI (IAS Zone cluster) if the sensor is enrolled? Also, please check deconz REST API if the state changes there.

geelenbert, 2020-01-29:
[image]

[image]

SwoopX, 2020-01-29:
I'm afraid the screenshots are not much of a help since they neither show the read attributes of the IAS Zone cluster nor answer the question, if you've checked the REST API.

geelenbert, 2020-01-30:
[image]

I paired a second one.
Looks the same. I read the IAS zone attribute. 

Do you think the device needs to be configured ? Or should the motion sensor work all the time ?

I'm not sure how to test the rest api. Should a debug log help ? If so, which log ?

SwoopX, 2020-01-30:
Thanks for the screenshot, that one is workable. It is strange that it hasn't been enrolled automatically (which should have happened). I can check the code once mor to see if I've overlooked something.

You can have a look here regarding API usage: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2373#issuecomment-578511439

geelenbert, 2020-02-03:
Did you have a momment to check if there was an issue in the code ?

I'm not technical enough I think to enroll it manually

SwoopX, 2020-02-04:
Yes, but haven't spotted anything suspicious so far.

> I'm not technical enough I think to enroll it manually

This is really just like 3 clicks and typing the MAC address of your conbee/raspbee in there. Looks worse than it is. I'm dependent on you with this since I don't have that device, otherwise I cannot really help.

geelenbert, 2020-02-09:
It was a bit of a struggle, but i finally found the steps that need to be taken.

By putting the mac address into the field the status changed to enrolled.
Home assistant does look like it reports changes of motion state.

So for now it looks like it is working!
[image]

SwoopX, 2020-02-09:
That doesn't sound too bad, great. Say, do you get reasonable battery readings from your device? My feeling would say no.

geelenbert, 2020-02-11:
I don't get any battery readings. 
Is there any way to check or improve this ?

SwoopX, 2020-02-11:
Already prepared an update.

geelenbert, 2020-02-11:
Cool!
Let me know when I can test it.

SwoopX, 2020-02-11:
Available already in my repo, so go for it.

Mimiix, 2020-06-05:
It seems this issue is resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix

geelenbert, 2021-01-04:
I dusted off this project again and i would like to mention that the device works out of the box.
Motion, 

Motion, temperature, lux and battery level:
[image]

Not sure yet if the battery level indicator is realistic, but 90% for a device that has been laying in the storage for some time seems reasonable.

One issue i'm having tough is that Phoscon does not show all these devices.
I have now added 7 of these sensors:

Deconz:
[image]

Home Assistant shows this: 
[image]

Phoscon:
[image]

The problem is that Phoscon only shows 4 of the 7 devices. I can use Phoscon to automate lights and rename the devices. 

Here is the node info per node:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

From my perspective, i don;t see any difference between the devices being shown in Phoscon, and the ones that are not. 
Removing and adding them again results in the same result.

Any idea what is going on ?

SwoopX, 2021-01-04:
I'll have a look later. Anyway, thanks for confirming!

geelenbert, 2021-01-07:
@SwoopX any change of pointing me in a direction ? 
i want to integrate the sensors, but i don't want to change the entities further on.

SwoopX, 2021-01-07:
This one drowned, sorry for that. Can you please read the IAS cluster attributes of both. I'd assume one is not enrolled.

geelenbert, 2021-01-08:
This is from one that is not detected:
[image]

This is one that has been detected:
[image]

I think they are enrolled successfully. When i started this thread, the devices did not enrol at all, so somewhere along updates of deconz or my system this looks fixed. Now they do not show up in Phoscon yet.

SwoopX, 2021-01-10:
Hm, that's odd. You could try entering sensor search in Phoscon again and trigger one of the problematic devices my provocing motion. In addition to that, it might be a good idea to push the physical button on the device.

Alternatively, you may want to try removing the batteries and re-inserting them while searching for sensors.