dresden-elektronik/deconz-rest-plugin #2290: Cannot log in to old web app

ginkel, 2020-01-06:
[image]

When attempting to access the old web app (either through /login.html or via the "Help" link), I am redirected to the Phoscon login page (/pwa/login.html) after entering my credentials ("delight" / <password> at /login.html) or clicking the link. Clearing cookies or using private browsing mode does not help.

Browser: Version 79.0.3945.88 (Official Build) Arch Linux (64-bit)

Smanar, 2020-01-06:
Hello, and with "index.html" ?

ginkel, 2020-01-06:
Same result. I briefly see the old app's toolbar and background when I get a redirect to the new app (/pwa/login.html). The redirect most likely happens from Javascript as I don't see any 30x issued within the Developer Tools' Network View.

brusarp, 2020-01-08:
Same problem for me...
Pi3+ with Raspbee
HA 103.6
Deconz 5.0
The problem appears after Ingress compatibility introduction...

Tony-Munich, 2020-05-03:
I wanted to post here since I had the same issue and finally managed to solve it with help from @Mimiix on the deCONZ Discord channels.

I'm running HA v109.2 supervised. When I tried to access the old web app from the link in the help I could see it for a split second then redirected back to the normal login. What I discovered was that I never set up any of the ports in the config for the deCONZ add-on. From the add-on page, at the very top is "Configuration". There are two section, config and network. In network, I had all three ports as null. I changed the first and last ones to 40850 and 8080 (**not 8081**), Restart HA. I still get the redirect issue but I can now use the direct link which is http://yourHAip:40850/login.html. This will bring you to a login screen that needs a username and password. The username will be "delight", the password is the password you use to log in to Phoscon ("delight" was the default here too but you likely changed this in the initial config).

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

@Tony-Munich Your most welcome ;)

closed 2020-06-06 by Mimiix

silfax, 2020-09-04:
same problem here, nothing seems to work. menu bar of the old app flashes briefly then i'm redirected to the login page. any advice? thanks.

Mimiix, 2020-09-06:
@silfax Please open a own issue 😄