dresden-elektronik/deconz-rest-plugin #2291: Hue Go did not connect

Piccar, 2020-01-06:
I tried to connect the the Philips Hue Go to the conbee II stick via the Phoscon Web-App. I tried multiple time to reset the Lamp(hold the button ca. 10sec until blue flashes) even drain all the Power from it. But the Web-app could not find it. Since the Website marked it as compatible I made myself a goaround: The only way for me to connect is was via the android app hue essentials. Once I was logged in the Lamp was found via Phlips hue bluetooth in secounds. Maybe somebody needed this tipp(Or it could be added in the wiki?)

Thomas-Vos, 2020-01-20:
Adding a light via Philips Hue Bluetooth connects it directly to Hue Essentials via Bluetooth. This means it is not connected to deCONZ and the light can only be controlled from Hue Essentials.

Piccar, 2020-01-20:
Strangely this method was the only way the Hue Go would connect to deCONZ. Im now able to control it within the deCONZ Controlpanel without the Android App. The App btw is removed from the devices and so could not be running in the background.

Piccar, 2020-01-23:
I connected the App to Phoscon through the 3rd Party Option.

Gizmo-Ger, 2020-02-29:
Same problem on my end as well. Unfortunalty I can´t get it to work with deCONZ. I can control it via Hue Bluetooth app and also Hue Essentials but not with deCONZ. Do I need to factory reset the Hue Go in order to get it recognized?

Thomas-Vos, 2020-03-01:
Could you try holding the button on the Hue Go for about 10 seconds until it starts flashing? This will set it to a pairing mode. Then try to conmect it with deCONZ.

Gizmo-Ger, 2020-03-01:
Thanks for the tip, Thomas. Unfortunalty I had no luck. deCONZ just won´t recognize the Hue Go.

Piccar, 2020-03-01:
Have you tried to connect the Hue Essentials app to deconz via the Third Party option and the connect the Hue Go to the app. This was the only way for me. Once the Hue Go paired to the App, deconz added this lamp to the known devices. Worked also on the tradfri lamps.

Gizmo-Ger, 2020-03-01:
Thanks Piccar, paired Hue Essentials to deCONZ and added the Hue Go Bluetooth in the app. I can control now the Hue Go via Hue Essentials app. Within Phoscon I did a new lights search but the Hue Go just won´t show up.

Piccar, 2020-03-02:
Strange.. in my case Phoscon added the Hue Go to the known devices when i connected the Lamp in the Hue essentials app. Then i deleted the App from my phone and its works like a charm. So do not give up!

Gizmo-Ger, 2020-03-02:
Naw, I won´t give up. Some other funny thing just happened. Now an unknown light device showed up which seems to be the ConbeeII
[image]
[image]

 itself.