dresden-elektronik/deconz-rest-plugin #2292: Add support for the Aurora AOne Double Socket UK

SamB-GB, 2020-01-06:
https://aone.auroralighting.com/ProductDetails/AU-A1ZBDSS

These currently add as a light and can be controlled on/off. However I can't see any power readings and using the physical on/off buttons there seems to be a delay of around 20 seconds before HA sees the state change.

[image]

[image]

ebaauw, 2020-01-06:
I would expect two lights, one for each endpoint.

The delay is caused by deCONZ polling the device; you need to setup attribute reporting for the _OnOff_ attribute in the _On/Off_ cluster (on each endpoint) so the device sends an instant notification to deCONZ when you change it using the physical buttons.

If you read the attributes of the _Electrical Measurement_ cluster, some values should appear.  I wonder if it really measures both channels independently; that would be a first.  Again, setup attribute reporting to get automatic updates of these.

If these values make sense, it might be opportune add support to the REST API plugin for one or two ZHAPower sensors (in addition to the lights).  In that case, please read the attributes from the _Basic_ cluster and post a screenshot of that, together with one of the attributes of the _Electrical Measurement_ cluster.

The spec says the LEDs can be configured high, low, or off, but gives no clue as to how.  Most likely a manufacturer-specific attribute (lacking a manufacturer-specific cluster).  I guess some-one would have to sniff the traffic between the socket and their hub to figure out how.  Or maybe just through a physical key combo?

SamB-GB, 2020-01-06:
Sorry I am a total noob at this so not really sure where to start.

So for the on/off cluster where do I setup the instant notifications? do I need to add/change something in the on/off cluster?

ebaauw, 2020-01-06:
You need to setup a binding from the _On/Off_ cluster to the first endpoint on the ConBee/RaspBee.   Then, you need to double press the OnOff attribute and write the reporting configuration attributes.  Typical values are 1/300/-.  See the deCONZ user manual (from the _Help_ menu in the GUI), sections 8.1 and 7.2.3.

SamB-GB, 2020-01-06:
Sorry to be a dumb head but when you say the first endpoint you mean the first Zigbee device that was added to the mesh right?

ebaauw, 2020-01-06:
In your screenshot, you see two endpoints for the socket, each with a protocol (ZHA), device type (Power Outlet), and a list of clusters.  If you press the right round button on the (blue) node for the RaspBee / ConBee, it will also display two endpoints.  Drag the first to the _Destination_ drop field and drag the _On/Off_ cluster of the first endpoint on the socket to the _Source_ dropbox.  Press _Bind_.  Then repeat for the _On/Off_ cluster on the socket’s second endpoint.

SamB-GB, 2020-01-06:
Ah got it now thanks, it's instantly reporting in HA now which is great thanks for the help.

Next I need to figure out how to get the energy readings from it. In home assistant this device is listed as a light (as it goes into the light group) but it doesn't show any energy readings from what I can see.

ebaauw, 2020-01-06:
Voltage looks legit, I doubt that it’s reporting 4A and 0W.  Could you connect a known load (water cooker, hair dryer) and see what it reports.  It might take some time before it takes a measurement, so be patient.  Make sure to read the attributes as well, in case the reporting hasn’t yet been setup, or has quirks.  You want to figure out what scale these values are in (A, 0.1A, 0.01A, mA).  Also check what the other endpoint is reporting.  Then switch the load to the other socket and repeat.

SamB-GB, 2020-01-06:
So with an up to 2000W hairdryer plugged in 240V (UK)

Hairdryer Plugged in socket one
[image]

Other Socket empty
[image]

Hairdryer plugged into socket two
[image]

It looks to me as though its.

RMS Voltage (V)
RMS Current (mA)
Active Power (W)

SwoopX, 2020-01-07:
@ebaauw 

>  Most likely a manufacturer-specific attribute (lacking a manufacturer-specific cluster). I guess some-one would have to sniff the traffic between the socket and their hub to figure out how.

Shouldn't we be able to see that with debugging enabled? The corresponding DB UPDATE lines should contain all the clusters from simple device descriptors, right?

SamB-GB, 2020-01-07:
It seems to be the "Level Control" cluster that determines the brightness of the LEDs

[image]

ebaauw, 2020-01-07:
> Hairdryer plugged into socket two

Is that the cluster on the other endpoint?  And did the first endpoint report 0W again?

> It seems to be the "Level Control" cluster that determines the brightness of the LEDs

Did you actually try that?  That would be very odd and a deviation from the standard; the _On/Off_ and _Level Control_ clusters on the same endpoint should work together (e.g. setting level to 0 sets onoff to false).  It’s not uncommon for plugs or outlets to expose a _Level Control_ cluster, so they can be controlled from wireless switches that send a _Move with On/Off_ command instead of _On_ or _Off_.

SamB-GB, 2020-01-07:
> Is that the cluster on the other endpoint? And did the first endpoint report 0W again?

Yes.

> Did you actually try that?

Yes

closed 2020-05-13 by stale[bot]

incarvr6, 2023-01-04:
I can see that it is closed, but it is worth a try.
How about energy measurements, is it really missing un this plug?