dresden-elektronik/deconz-rest-plugin #2294: ConBee not seeing anything

jcaron23, 2020-01-07:
Hi all,

I bought a ConBee quite a while ago. When it didn't work I set it aside, but now I'm back trying to make it work and I haven't had much success.

I've tried using it on my Mac with deCONZ running in Win7, Win10 or Ubuntu VMs via Parallels. I've tried using it on a Raspberry Pi 4 B with one of the pre-built SD cards images. I never can see any device (I've tried an old Hue bulb, a Hue Motion Sensor, an OSRAM SMART+ plug and a Trust motion sensor). I've tried plugging in and out, switching on and off, resetting, many many times, nothing is ever detected.

The ConBee correctly shows up with the latest firmware (26330500). In deCONZ it shows as "In Network", and I have the node for the ConBee with a flashing blue dot on the left. But whatever I try, I never see any other devices.

I have also tried using the Mozilla IOT gateway, and the result is the same. The ConBee is correctly detected, but no other device is ever detected during scans.

I suspect in the end it's just the ConBee's radio which is dead, but is there anything else I could try to confirm this? Is there anything that is stored in the ConBee itself (rather than in deCONZ) which I could have messed up at some point and which I could try to reset?

Any hints welcome!

raulfunkie, 2020-01-07:
Going to chip in and ask if you have anything else plugged on any of the USB ports of the Pi. Yesterday, I had trouble getting my ConBee to recognize any device (I tried with Vibration Sensor, Temp Sensor and Door/Window Sensor), no sensor was being detected.

At a last resort, I unmounted an external drive I have plugged in and magically it all started working. The three sensors were detected almost instantly and I was able to set up. My external HD is/was connected in a USB3.0 Port (I'm using a Pi 4). After pairing the devices, I re-connected and mounted the drive and everything seems to be working well.

Maybe the ConBee requires some higher energy to properly activate the scanning mechanism? I'm just guessing here but it's something worth trying.

jcaron23, 2020-01-07:
Not much else connected, a keyboard, mouse and touch screen right now. Tried disconnecting all of them, no change.

I tried using zshark and I see beacon requests received by the dongle, so the RX radio part seems to be working at least.

Currently monitoring using a CC2531 and Wireshark, and can still see the beacon requests from the various Zigbee devices I have, but nothing from the ConBee. Seems like either the radio TX part of the dongle has an issue or I'm missing something...

jcaron23, 2020-01-09:
OK, so I received a brand new ConBee II yesterday, and it "just worked™" right way :-)

So I guess the original ConBee indeed has a hardware problem. Can still use it as a sniffer, so that will be its job.

closed 2020-01-09 by jcaron23

joebloggins, 2020-02-07:
Out of curiosity, did you try the new ConBee with Parallels?

jcaron23, 2020-02-17:
@joebloggins I hadn't. I tried it, and the result was less than stellar, with the ConBee II remaining in "Not in Network" state most of the time. At some point on Friday it got in "In network" state, though I have no idea why or how. I tried this morning again, and could never get it to connect despite numerous restarts of the software, reinstalls, reboots, resets, and whatnot.

I have no idea, however, if the issue comes from Parallels or from the fact that I connect the ConBee to a USB-A port on a USB-C dock (given it's connected to my Mac). There are apparently lots of issues related to exact how and where you connect the ConBee, so the problem may come from that rather than Parallels.

I'm honestly a bit underwhelmed by the whole experience. Between the first ConBee which I never got to work, the second one which loses all connections on a reboot and doesn't work in Parallels and/or with a USB-C dock, it all seems very very fragile. I had projects I was hoping to use the ConBee for, but at the moment I most definitely can't count on it for them.