dresden-elektronik/deconz-rest-plugin #2295: Deconz-Gui on a headless Raspberry

1onar, 2020-01-07:
Hello,
i got a Problem running deconz-gui on my headless Raspberry.

I got RealVNC running this way [https://www.raspberrypi.org/documentation/remote-access/vnc/](url).
But when i try to start Deconz-gui this error occurs:

`Jan 07 11:22:27 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Jan 07 11:22:31 raspberrypi deCONZ[613]: Invalid MIT-MAGIC-COOKIE-1 keyqt.qpa.screen: QXcbConnection: Could not connect to display :0
Jan 07 11:22:31 raspberrypi deCONZ[613]: Could not connect to any X display.
Jan 07 11:22:31 raspberrypi systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Jan 07 11:22:31 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
`

SwoopX, 2020-01-07:
Do you get a working VNC session? The error output states that you have issues with the .Xauthority file. Do you use seperate users for deconz and vnc connection?

Smanar, 2020-01-07:
If you have an headless installation, it can't work, you need a desktop instalation, VNC is just here to connect to the desktop interface.

1onar, 2020-01-07:
Yes i get a working VNC-Session on 192.168.178.24, also on 192.168.178.24:0 and 192.168.178.24:1 with realvnc (i dont know the difference tbh).
I tried to open up the webbrowser and it doenst show up also in realvnc.

I dont think so. Both running on user root.
I am a beginner, sorry.

SwoopX, 2020-01-07:
> Yes i get a working VNC-Session on 192.168.178.24 and 192.168.178.24:0 with realvnc

So you do get a desktop?

> Both running on user root.

Unless you haven't explicitly changed anything, you're not root but user pi. Regardless, I'd try to run deconz from command line as follows:

```
sudo systemctl stop deconz-gui
sudo systemctl stop deconz
/usr/bin/deCONZ
```

1onar, 2020-01-07:
Yes, i got a desktop. But i cant open the web browser or deconz-gui.

Did both commands and thats the result:

`root@raspberrypi:~# sudo systemctl stop deconz
root@raspberrypi:~# sudo systemctl stop deconz /usr/bin/deCONZ
Failed to stop usr-bin-deCONZ.mount: Unit usr-bin-deCONZ.mount not loaded.
`
I tried to open Deconz afterwards. Still nothing happens or appears.

1onar, 2020-01-07:
> sudo apt-get update
> sudo apt-get upgrade

Fixed it. But thank you for your effort @SwoopX

closed 2020-01-07 by 1onar